adafruit/pi_video_looper #4: Check for file change in directory mode

thecrazy, 2015-04-09:
I see you already thought about this one,

Why not use stat on the folder?

Store the output of "stat -c %y /path/to/folder/" and compare it at every loop, if it changes you know the content of the folder was modified and it hardly takes any time:
 
real	0m0.043s
user	0m0.010s
sys	0m0.000s

Now, I know what your thinking... but I don't know KungFu... er I mean Python and when I tried modifying a simple text string in usb_drive_mounter.py to change the message that is displayed on screen saying "insert usb key etc..." the video looper wouldn't start anymore... So not much luck there :(

Had to undo the change to get it going again.

tdicola, 2015-05-04:
Oh interesting idea--yeah that would be a good way to detect when files are added.  I don't think I'll be able to get to making the change myself, but if someone wants to take a stab at making the change and submitting a pull request, go for it!  Thanks again for suggesting the idea!

closed 2015-05-04 by tdicola

thecrazy, 2015-05-04:
Turns out it wasn't as reliable as I thought, so instead I do a simple ls -la on the directory and compare the output. Works like a charm, every time but i have it implemented using a bash script to assist cause im not to sure how to do it in python. It even works with paths that can get dismounted like external drives, network mounts, etc...

thecrazy, 2015-05-04:
It makes the usb mode redundant.

matteoraggi, 2017-01-01:
I am interested to check for file changes in directory mode too. any ideas? What about the cost to add this feature paying a developer?