adafruit/pi_video_looper #5: 20s Single Video

flexvisual, 2015-06-03:
I have a couple of video clips that are 20s long with a time code burn. When I place them on a flash drive with multiple clips they play through. The issue is when they are the only clip on the drive, they loop at they loop at the 16s point. The Product is a Raspberry Pi 2B. It has had every update except firmware. Any ideas on how to correct this?

I am using omxplayer, It has not been tested on hello_video.

Thanks, Chad

tdicola, 2015-06-08:
Yeah I noticed similar behavior and it looks like a limitation of OMX player.  When it loops it cuts off a bit of data that it has buffered and it's much more noticeable with short videos.  

If possible one thing you could do is extend the clips a bit to account for the clipping.  Another option is to adjust the buffering that omxplayer player uses to see if less buffering has less clipping.  You can actually change this in the configuration file, if you edit /boot/video_looper.ini on the SD card find the line that looks like this:
```
# Any extra command line arguments to pass to omxplayer.  It is not recommended
# that you change this unless you have a specific need to do so!  The audio and
# video FIFO buffers are kept low to reduce clipping ends of movie at loop.
extra_args = --no-osd --audio_fifo 0.01 --video_fifo 0.01
```
Try bumping the audio_fifo and video_fifo values down to even smaller values like 0.001 (unfortunately I don't think you can provide 0 or disable the buffering entirely with omxplayer).  Save the file and try running again to see if that reduces some of the bits cut on the end of looping.

andreas-andersson, 2015-08-19:
I have the same issue with a 10s mov file, cuts at 8s. Sadly changing the fifo values down to even .0001 did not help, no change. Have yet to try hello_video...

tdicola, 2015-08-19:
Ah, unfortunately it looks like a limitation of omxplayer swallowing the last 2 seconds in its buffer.  Your best bet to work around it is to pad the video with a couple extra seconds of blank frames.

bf5man, 2015-08-30:
This [simple fix in the omxplayer main project](https://github.com/popcornmix/omxplayer/pull/340) appears to solve this issue. I don't have what it takes to test it but maybe someone could try it in the Adafruit/Omxplayer repository?

bf5man, 2015-08-30:
I managed to compile and install my first program, the Adafruit/omxplayer version with the fix in it. It works great. I get a seamless looping without a trimmed end. Since I'm not mastering Github, I'll let someone else push the modification in the official repository.
Thanks tidicola for this super useful tool!

tdicola, 2015-09-01:
Oh wow thanks for tracking down and testing out the fix.  I'll keep this issue open so we can look at merging in that version of omxplayer.  Since they've actually merged in our original fix that required the fork I think we can probably just swap the video looper over to start using the normal omxplayer builds and pick up this and other fixes going forward.

Lyptik, 2015-09-28:
Hello Tony, I saw that popcornmix did add on their repo the fix, did you keep up the adafruit omxplayer to that ?

Lyptik, 2015-09-28:
I did test with latest version of omxplayer from adafruit and the video looper and it seems now to be ok. I could not test frame wise (using test.h264 given with the raspi). Can tell than more than 3 seconds were not swallowed !

bf5man, 2015-09-28:
Lyptik, if you still doubt some seconds might be trimmed from your video, you might want to try this version of [Omxplayer I've compiled here](https://github.com/bf5man/omxplayer/releases). As far as I know, it doesn't trim the video at all.

Lyptik, 2015-09-28:
Thanks ! Actually it really seems to work with what is on this repo right now, I will stick to that for now. (Raspberry 1 B+)

stevelewis33, 2016-08-22:
I have just installed the pi_video_looper software on a brand new Pi 3 Model B (excellent software, much easier than the manual solution I have used before). I left everything at default settings and inserted a USB memory stick containing an mp4 file, the first iteration of the loop worked flawlessley but every iteration from then on cut the last couple of seconds off the video. It seems this issue has returned, or is still not properly fixed. 

Thankfully the video we are looping does not require sound so we changed the video player to hello_video in the .ini file. I have a small amount of past experience using hello_video and I knew that it could support .m4v files exported directly from our Adobe Premiere Pro and Adobe After Effects software. However this file type **was not recognised by the pi_video_looper** software when inserting the USB. 

I read your guide and on [this page](https://learn.adafruit.com/raspberry-pi-video-looper/hello-video) which instructs to use ffmpeg to convert files to a .h264 file. Upon further examination of the .ini file I found the section for specifying file types for hello_video. I added .m4v to this list in addition to .h264 and it now works flawlessly with our m4v files. 

As there doesn't seem to be any easy way to export raw .h264 files from the Adobe software can I suggest you add m4v to the default .ini file and perhaps add a note to the above linked instructions page so that Adobe users are aware they don't need to convert files using ffmpeg?

Lyptik, 2016-08-23:
pretty cool @stevelewis33 !

tofuSCHNITZEL, 2019-08-24:
currently I cannot reproduce this issue. my 10s file plays fine from beginning to end (even with omxplayer)

closed 2019-08-24 by tofuSCHNITZEL