adafruit/pi_video_looper #6: hello_video not working

jardimsonoro, 2015-06-06:
On a fresh install on raspbian, on B+ model, the option to use Hello_Video is not working for me.
After boot, I can see the terminal window flick 5 times (exactly the number of .h264 videos I have on videos folder) As if the player tries to play 'em but fails instantly.

Omxplayer option works fine (with .mp4 videos)  but I am looking for gapless transition between videos.

And just to make sure, I can call hello_video.bin from command line, and play every single .h264 video there is on the videos folder, so they are proper files, and hello_video.bin is compiled and working. It gotta be something very easy to fix.

tdicola, 2015-06-08:
Hrm that's odd--just to check try running one of the videos with the same command line parameters as the video looper.  For example if you have a foo.h264 it will call it like:
[code]
hello_video.bin --loop foo.h264
[/code]

Also the USB mounting logic will mount USB sticks under a path like /mnt/usbdrive0, /mnt/usbdrive1, etc.  Try to run hello_video.bin with it pointed at a file on a USB drive to help rule out any odd permission issue.

jardimsonoro, 2015-06-08:
Yes, command line works for every single h264 i have. And the loop is seamless (for a single video from cmd line call)
I am using local directory.
I dont understand why hello_video wont start the video folders at boot..
Omxplayer works bur I really need seamless transition between videos inside a folder.
Thank you Tony.

tdicola, 2015-06-08:
Ah thanks for confirming hello_video works when run with the file's separately.  It sounds like perhaps there's a problem with the video looper reading the files from the the directory.  One quick think is to make sure the directory isn't private or hidden in some way--although the video looper by default runs as root using the tool supervisord so it should have access to everything.  Another thing to check is the log it keeps, you can actually watch the output in real time if you log into the pi (using SSH) and run the command:
```
sudo supervisorctl tail -f video_looper
```
Try grabbing the output of the log to see if it has any errors.  It should log on startup how many movies it found and then log each time it plays one, like:
```
Found 5 movies.
Playing movie: /path/to/movie.file
Playing movie: /path/to/movie2.file
...
```
Also tail the standard error output from supervisorctl to see if anything is there, you can run:
```
sudo supervisorctl tail -f video_looper stderr
```
Let's check if there's more info in one of the logs that can help explain why the movies aren't playing.  Thanks!

jardimsonoro, 2015-06-09:
Hello Tony, 

Thank you for you support on this.

The first log seems ok:
_____________
Found 2 movies.
Playing movie: /usr/videos/1.h264
Starting Adafruit Video Looper.
Found 2 movies.
Playing movie: /usr/videos/1.h264
______________

Now on the second log there are some warnings:

# supervisorctl tail -f video_looper stderr
==> Press Ctrl-C to exit <==
ooper.py", line 252, in <module>
    videolooper.run()
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_Video_Looper-1.0.0-py2.7.egg/Adafruit_Video_Looper/video_looper.py", line 218, in run
    self._player.play(movie, loop=playlist.length() == 1, vol = self._sound_vol)
TypeError: play() got an unexpected keyword argument 'vol'
Traceback (most recent call last):
  File "/usr/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/usr/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_Video_Looper-1.0.0-py2.7.egg/Adafruit_Video_Looper/video_looper.py", line 252, in <module>
    videolooper.run()
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_Video_Looper-1.0.0-py2.7.egg/Adafruit_Video_Looper/video_looper.py", line 218, in run
    self._player.play(movie, loop=playlist.length() == 1, vol = self._sound_vol)
TypeError: play() got an unexpected keyword argument 'vol'

Lyptik, 2015-06-09:
Hello Guys,

This problem seem to be related to something I did implement. Either something went wrong in the pull request or I did leave something behind anyway.

I guess you have to remove the « vol » argument part which is related to omxplayer playlist and not this player as it cannot read sound.

jardimsonoro, 2015-06-09:
Hi Lyptik.
Can you help me with some basic explanation on how to:  "remove « vol » argument" .  
From what file? Do I have to reinstall the hello_video.git ?

Lyptik, 2015-06-09:
Hello,

Quickly, from what I see, here is a guess you can try (I do not have the architecture in mind) :

"
self._player.play(movie, loop=playlist.length() == 1, vol = self._sound_vol)
TypeError: play() got an unexpected keyword argument 'vol’
"

Try to replace that : self._player.play(movie, loop=playlist.length() == 1, vol = self._sound_vol)

by that :

self._player.play(movie, loop=playlist.length() == 1)

or find a way to make the vol arg optional.

If you want a quick solution, you can also simply revert back in the git history before there was the « volume » feature, which is in one of my commit. This should fix your problem.

Sorry I do not have more time to look into it now.

Best,

David C.

jardimsonoro, 2015-06-09:
Ok, forgive my newbines..

Can I just edit a file without having to recompile anything?
What is this , and where is it located?

so i can:
"replace that : self._player.play(movie, loop=playlist.length() == 1, vol = self._sound_vol)
by that :
self._player.play(movie, loop=playlist.length() == 1)"

Lyptik, 2015-06-09:
Hello Jardim,

Hmmm, Tony would be of better advise than me, I remember that you have to trigger some scripts to move the file to the correct place.

To answer your question, no compilation because it is python, but some scripts to move the file to the correct place ;)

If you want a quick fix, I strongly advise to follow what I describe in my last mail.

This means : Go back into the history of git to the commits before mine.

If you don’t know how to do it, go into a terminal in the folder where is the code and type :

git checkout 6cd7a794a06138006149647a20bf2570b6785e36

This should make your code go back into a correct state.

Then trigger the installation script to update everything.

Restart your raspbeerry

David C.
david.alexandre.chanel.ext@gmail.com

tdicola, 2015-06-10:
Oh ack, yeah it looks like the merge to add the sound support has a small issue with calling the hello_video player (even though hello_video doesn't use sound, it will still get called with the volume param but fails unexpectedly).  I just commited a quick fix, can you grab the repository and install it again jardimsonoro?  Thanks for grabbing the logs as those helped to fix the problem very quickly.

jardimsonoro, 2015-06-12:
Hello Tony.

This new version boots with hello_video, 
But at the end of the first video freezes on last frame and dont play the next video on folder.
I can confirm all they are all .h264 and plays from command line.
This time I get no useful information from the logs. 

"tail video_looper" returns these lines only:

Starting Adafruit Video Looper.
Found 3 movies.
Playing movie: /home/videos/1.h264

and "tail video_looper stderr" returns nothing more than:

Press Ctrl-C to exit

Lyptik, 2015-09-28:
Hello Jardis, did the fix fixed your problem ?

Lyptik, 2015-09-28:
Hello Toni, I confirm that before the fix it does not work and after it does, you can safely close this thread :)

tdicola, 2015-10-05:
Sounds great, thanks for the followup!

closed 2015-10-05 by tdicola