adafruit/pi_video_looper #9: playlist

rezervoar2, 2015-08-20:
Instead of scanning a directory for videos, could you make an option to input a playlist in whatever format. For the thing I need, in one sequence, I'd replay one of the videos several times, and having several copies of the same video is not so cool. I'd like to fork it and do it myself, but I haven't used python before, so I'd need a couple of days at least to get a hang of it and I haven't got them. Rather than making an option, could you tell me the idea here? I've read the code a bit, found where it scans for files, but getting just the file list without the rest of the parameters you're getting from the files themselves is not gonna work.
Thanks in advance

tdicola, 2015-08-21:
Oh good question, yeah unfortunately there's no playlist support right now.  One thing you could do is setup a symbolic link to 'copy' the file that plays multiple times without actually taking up all the space of physical copies.  It's an odd hack but should work in the short term.

Longer term though yeah I'm curious is there a standard playlist format that would be easy to use?  Perhaps VLC or similar have a playlist format we could look at adding?

mug3n, 2015-08-22:
Yeah I thought about something like that as well.
M3u is a standard indeed. Its format is simple as well. Though it would still need to scan the files from the playlist to get the needed data I guess. I will try to do it myself and post the results if I manage it.
I wouldn't call it a bug though, rather a feature :D
Thanks for the quick answer,
Cheers

tofuSCHNITZEL, 2019-08-24:
@raegen version 1.0.3 introduced the ability to control the repetition of files. the order of the files can be controlled by the filename (the internal playlist is sorted alphabetically)
maybe this helps achieving what you need

closed 2019-08-24 by tofuSCHNITZEL