adafruit/pi_video_looper #11: Player reads but doesn't play

Lazyfattree, 2015-10-02:
I've got image while booting, it reads my file in the countdown. But when it should play the screen stays black.

tdicola, 2015-10-06:
It sounds like there might be a problem with the video player and the file you're trying to play.  Check out this reply for info on how to examine the log files that the video looper generates: https://github.com/adafruit/pi_video_looper/issues/6#issuecomment-110120008  Check the logs and let's see if there are any useful errors which could help troubleshoot the problem.  Thanks!

mfgeorge, 2015-12-17:
Hello Tony, 
I am currently having this problem as well. I am sure that my video is h264 encoded as I have re-encoded it in MEncoder using:
mencoder myvideo.mp4 -o myh264.avi -ovc x264 -oac pcm

I went to #6 comment and saw the volume issue that was occurring. I have checked out the version suggested in that forum, and yet am still having this same issue. 

Currently in the process of getting logs now, but first I will be getting some sleep for the night.

Thank you for your great project and your help

mfgeorge, 2015-12-17:
Ok... didn't go to bed, but figured it out.

The problem was not having omxplayer installed. I had assumed it was distributed with this project. A simple apt-get fixed this :). 

Sorry for commenting too soon. Hopefully this can help other new learners like me.

Lyptik, 2015-12-17:
"Ok... didn't go to bed, but figured it out." That's the spirit dude :)

This is rather strange as I think the install script installs omxplayer, did you use it ?

mfgeorge, 2015-12-17:
I did use the install script and I checked out the git version with the
volume control disabled. Tried running the video looper but just got a
black screen after the count down. I removed the boot file so that I could
get back into the raspberry pi terminal on boot.

Then I figured it would be best to understand the video player itself so I
installed it and ran it on a file to make sure my file was correctly
encoded. It ran perfect first try, then I reinstalled the video looper, and
everything worked right away.

Hopefully this can help someone in the future.

closed 2019-08-24 by tofuSCHNITZEL