adafruit/pi_video_looper #13: Exfat usb read

Lazyfattree, 2015-10-05:
I've followed the lines of htpcguide to properly mount an exfat usb on a raspberry pi. 

But when I install the video looper. The raspberry flickers between a black screen and the os ui. 
If I then pullout the usb it shows the screen to put the usb in to play a movie.

So it reads the usb but from there on out it doesn't work. Tried HFS+ , NTFS and none works.

I need to get a file bigger than 4.3 gig on the video looper.

Lyptik, 2015-10-05:
Just jumping on it quickly,

Since it is linux, you should try to format your key using eiether ext2, ext3 or ext4.

This should I think handle big files, and it will be readable by your raspberry.

m-p-3, 2017-12-04:
On my end, I seem to be able to play a video file from an ExFAT formatted USB thumbdrive, and all I had to do was `sudo apt install exfat-fuse` (currently using Raspbian Stretch Lite).

tofuSCHNITZEL, 2019-03-22:
> sudo apt install exfat-fuse

thanks for the tipp
I included this in v1.2.6 of my fork
https://github.com/tofuSCHNITZEL/pi_video_looper

tofuSCHNITZEL, 2019-08-24:
its now also implemented in v1.0.2 in this repository

closed 2019-08-24 by tofuSCHNITZEL