adafruit/pi_video_looper #14: Audio level output is low, how can I increase

Oldertarl, 2015-10-27:
Quick question before halloween where we are using the looper.
The audio feedback from the Py audio jack is low.

I read in the ini file that this can be adjusted. just no clue how.
# Sound volume output for the video player will be read from  a  file  near  the
# video files. If the file does not exist, a default volume of 0db will be  used

sound_vol_file = sound_volume

Do I need to create a file on the USB drive where my movie is that has a specific name and content to set the audio output level?
Please advise as this will be a key feature in our setting.

Many thanks in advance

bbberttt, 2016-01-13:
Any solution for this problem? got the same problem, but my volume is too loud. Thanks!

tdicola, 2016-01-14:
Oh no problem, sorry I should have added to that comment to clarify.  Here's the pull that added volume support if you're curious: https://github.com/adafruit/pi_video_looper/pull/1  The way it works is the .ini file defines the name of a file to look for that will have the sound level inside it.  By defaul the ini sets:
```
# Sound volume output for the video player will be read from  a  file  near  the
# video files. If the file does not exist, a default volume of 0db will be  used
sound_vol_file = sound_volume
```
So it will look inside the same directory as the videos for a file called 'sound_volume' (with no extension).  Inside that file should just be a single line with a sound level to pass to omxplayer (using the `--vol` option, it looks like it takes the value in millibels according to the usage info here: https://github.com/popcornmix/omxplayer).

So to set the sound level try creating a file called 'sound_level' (without quotes) inside the directory with videos (like on the USB key), and put inside the file a sound level to use like 100 (for 1 dB boost I believe).

closed 2016-01-14 by tdicola

Lyptik, 2016-01-14:
True,

Maybe the comment (my bad) should be better clarified that it is a millibel et not db value for a better understanding.

Best,

David C.

jjab2020, 2016-08-04:
Hi, first I would like to thank you about this work , can you help me to hide videolooper and keeping video playing or stopped and show desktop to run another application and return to movie to played again any idea to make videolooper hiding and keeping video thank you very much

Helise, 2018-04-18:
Hi, 

I've added a file in the usb directory called: 
sound_volume

In this file (without extension) I put the line:
omxplayer --vol -3000

But it does not change the volume.
Is the line wrong?

Thanks