adafruit/pi_video_looper #15: Monitor blank after 15min

Lazyfattree, 2015-11-13:
I've did a lot of test before I put this issue on here.

But following still occurs, this being that after 15 ish minutes the monitor on my raspberry pi goes blank.
The green light then stops blinking on the raspberry and only the red light is on. I can still connect to ssh.
Changed adapter, sd card, raspberry Pi. 

So my guess is that the screens goes black due to the monitor or due to the raspberry pi that can't get the screen back on the monitor. 

Help is needed :+1:

Lyptik, 2015-11-13:
To correctly isolate the problem you still have to :

- change screen
- change OS
- document yourself about the meaning of the green light (heart beat)

Lazyfattree, 2015-11-13:
I've changed the OS as well.
Also there are 30 other screens with the same setup that do work.

The green light is mostly irregular when playing the movies. But when the screen goes blank the irregularity stops.

So my guess is the screen to change. But I have to push this true the people who give me the assignment.

brennen, 2018-08-02:
This doesn't appear to be an issue with the code here.

closed 2018-08-02 by brennen