adafruit/pi_video_looper #17: scheduled playlist

baha00, 2016-02-02:
hi is it possible to ad more directory sources, for example /home/pi/videos runs from 12-00 to 13-00 an than from 1301-1400 run videos from /home/pi/videos2

tgld, 2019-03-14:
This post is old now, but I was just trying to do the same thing and figured out something that worked, so I thought I'd post for others...

The trick is that the call to video_looper can take an input argument of the video_looper.ini file you want to use, instead of using the default.

To start, the video_looper.ini file has the video folder path, so I created a copy of the .ini file, we'll call it video_looper_new.ini. In this copy, I changed the video folder directory to /home/pi/video1.

Since video_looper is called by supervisor, I had to create a new supervisor conf file in /etc/supervisor/conf.d
In this folder, create a copy of the video_looper.conf file. I called mine video_looper_new.conf.
Open the file and on the program line, change from video_looper to video_looper_new.
On the command line, add an input argument at the end of the line. So in this example, the end of the line will be:
Adafruit_Video_Looper.video_looper '/home/pi/video1'
On the autostart line, change the value to false. If this is true, then both videos will start playing when the device reboots, and it doesn't handle that well.
 
Now the original video directory can still be used as it was before, but the new directory can also be called:
supervisorctl start video_looper_new

If you do this, you do have to be careful not run 2 instances of the video_looper at the same time. To prevent this, every time I stop a video, I call a script that stops both instances video_looper to make sure nothing else is running.

bsiegrist, 2019-10-16:
Again someone with a similar problem but no solution to it. Maybe someone can help me out there...

What I want in the end: I need to loop video1 the whole day. The next day, video2 needs to loop the whole day. On day 3, video 3 needs to loop the whole day... You see where this is going.

Problem 1: I will not be in the same location as the raspberry in these days.
Problem 2: There's no Internet connection. So no chance to change the file via dropbox or similar.

So I thought of timing it with Cron and change the path every day (all videos fit on the sd of the raspberry). But since I'm a newbe I didn't manage to code this.

Any ideas?

tgld, 2019-10-16:
> Again someone with a similar problem but no solution to it. Maybe someone can help me out there...
>
> What I want in the end: I need to loop video1 the whole day. The next day, video2 needs to loop [...]

This is exactly what I had to do. I was scheduling 4 videos in advance to play at their proper time. Where are you getting stuck?

bsiegrist, 2019-10-16:
Sorry, maybe i understood your comment wrong. Don't you need to choose between the directories in your solution? I will not be there to give any commands... It needs to change automaticly.

tgld, 2019-10-16:
> 
> 
> Sorry, maybe i understood your comment wrong. Don't you need to choose between the directories in your solution? I will not be there to give any commands... It needs to change automaticly.

Yes, they are separate directories. I scheduled separate events in cron to turn the different videos on and off on the correct day.

tofuSCHNITZEL, 2019-10-16:
a very, very dirty (imho) trick would be to have multiple video_looper.ini files and in them you set different directories.
then you can override the default config path by using e.g. 
`python3 -u -m Adafruit_Video_Looper.video_looper /path/to/config.ini`

you can put this in the cronfile but also don't forgett to kill any running videolooper before starting the new one
you also should disable the autostart of the video_looper service (run disable.sh)

but be aware: as far as I know, a rasperry pi without internet to sync the clock will have a rather significent drift its internal clock and the cron job might not launch entirely accurate over long periods of time

bsiegrist, 2019-10-16:
Thanks for the input. I will make a test run!

tofuSCHNITZEL, 2023-11-16:
Maybe this could help you as well? https://github.com/adafruit/pi_video_looper/issues/195#issuecomment-1669486348

closed 2023-11-16 by tofuSCHNITZEL