adafruit/pi_video_looper #19: issue with setting window size to pass into omxplayer

master-chang, 2016-03-22:
So I have been trying to use your video_looper on a Square 960x960 Samsung monitor for art installation, and when I edit the video_looper.ini to try set the --win '0 0 960 960' arguments into the section for them they don't seem to pass over into omxplayer, and the video is squished vs play fullscreen properly.  

extra_args = --win '0 0 960 960' -- no-osd --audio_fifo 0.01 --video_fifo 0.01

If I use omxplayer alone to play a single video that is formatted at 960x960 with those arguments it displays the videos correctly at the correct aspect ratio and is not squished my on my monitor.
But when I play them using video_looper they get squished.

Been racking my head around it and would love some help with this issue, if at all possible, not sure if I am doing something wrong. 
Thanks.

Matsadona, 2016-09-06:
I have a similar problem when playing videos through the composite video output - and some content is outside the visible area. The config.txt overscan values doesn't have any affect.

The moive files played are 320*240 and approximately 10 pixels are lost in the top and left part of the screen.

ismotee, 2016-11-22:
I wonder. --win uses '' marks in arguments. There's a possibility to those marks mess up the python code. omxplayer.py passes on the arguments as it is to subprocess call. So, maybe something like this works:
extra_args = --win "/'0 0 960 960/'" --no-osd --audio_fifo 0.01 --video_fifo 0.01

NesbyBone, 2017-04-12:
I'm trying to get the video in the upper left corner of the screen, with a black backgound and I couldn't do it, would you give me a hand?

I run it manually with:

omxplayer -o both -b --win "0 0 300 220" video.mp4

How can I add it on your video_looper.py?

This is what I need:

[image]

Thank you so much in advanced

Hope to hear from you soon

elalemanyo, 2017-04-13:
@NesbyBone have you tried to add your new parameters at video_looper.ini extra_args parameter?
Check the result at omxplayer.py line 36. Let me know if it works.

Romankornfeld, 2017-04-13:
Mine looks like that.
extra_args = --no-osd --audio_fifo 0.01 --video_fifo 0.01 --win 149,449,224,524

elalemanyo, 2017-04-13:
And what happens? Should not be just: -o both -b --win "0 0 300 220"?

elalemanyo, 2017-04-13:
And works like this?

Romankornfeld, 2017-04-13:
yes

NesbyBone, 2017-04-19:
@elalemanyo and @Romankornfeld thank you so much both of you, it worked great!