adafruit/pi_video_looper #22: I suggest linking to https://learn.adafruit.com/raspberry-pi-video-looper/installation

bean5, 2016-04-13:
You could link to https://learn.adafruit.com/raspberry-pi-video-looper/installation either in project settings or in the README itself. Just an idea. Thanks for the project! Do you know of a way to do picture looping?

matteoraggi, 2017-01-04:
http://superuser.com/questions/211561/loop-pictures-from-a-folder-and-display-fullscreen-slideshow

joelhaasnoot, 2017-01-15:
Best solution to do pictures is to add a player that does pictures using fbi. I need this for a project soon, I'll see if I can do a pull-request.

brennen, 2018-07-25:
It's a good suggestion.  The README's due for a bit of an expansion; I'll link the guide when I'm doing that.

tofuSCHNITZEL, 2019-04-16:
I linked the guide in my recent README update. @brennen if you have additional ideas please create a pull request, thanks!

closed 2019-04-16 by tofuSCHNITZEL