adafruit/pi_video_looper #32: Change messages into another language

pittipuk, 2017-08-22:
Hi,
how (where !) can I change the messages like "Insert USB drive with compatible movie..." ?
I want to use the videolooper in another language.
I didn´t find the file, where I have to change the messages.
Thanks for your help!

reggae-shark, 2017-08-25:
Same question! I'd like to have no text

arturodr, 2017-09-08:
before installing change the file: 
```
Adafruit_Video_Looper/usb_drive.py
```
around line line 40 you'll see:

```python
def idle_message(self):
        """Return a message to display when idle and no files are found."""
        return 'Insert USB drive with compatible movies.'
```
Just change that function to return  your desired message.

whiteaden, 2018-05-02:
@arturodr this is great, the predicament I'm running into is breaking the instructions into multiple lines; `\n` doesn't seem to work, it just shows a ⍰ on my display instead of dropping stuff on another line.

I'm a scrub with python so maybe it's not a possibility in the way the idle_message is set up.

arturodr, 2018-05-16:
@whiteaden AFAIK idle_message is only used by a print() function at video_looper.py 

I have no way to test it right now but something lie this should work:
```python
def idle_message(self):
        """Return a message to display when idle and no files are found."""
        return """First Line
Second Line
...
Last Line"""
```

tofuSCHNITZEL, 2019-08-24:
I will close this for now. If anyone knows how to properly create translations for pythons scripts (maybe gettext?) feel free to create a pull request.

closed 2019-08-24 by tofuSCHNITZEL