adafruit/pi_video_looper #33: Run in RO mode

jonaslund, 2017-08-22:
Hi

Are there any instructions for running this in on a RO operating system? 
Any log files that need to be moved elsewhere?

Thx.

rikrikrik, 2017-08-22:
What do you mean RO operating system?

tofuSCHNITZEL, 2017-08-22:
Make the SD Card as much read only as possible so that the File System does not get corrupted when you unplug the rpi without shutdown - this is also I feature I need for one of my projects.
But there are plenty of tutorials of how to achieve this e.g.
https://hallard.me/raspberry-pi-read-only/

But as of now I have no experience which method works the best.

Lyptik, 2017-08-22:
Hello guys,

Check my fork, there is some useful scripts and indication about the 
readonly process. I have to warned you I still had some corupted SD 
cards with raspberries even with having a whole working readonly system.

jonaslund, 2017-08-23:
Yeah, read only mode. I used that guide, 
https://hallard.me/raspberry-pi-read-only/ 

But one done, omxplayer refuses to start as it can't access /tmp/ for storing the runtime variable. I hack-solved it now by putting the startup script in rc.local and just executing the python video-looper straight away as root... 

Hack solution, but works.

closed 2017-08-23 by jonaslund

phatpaul, 2022-09-12:
I was able to achieve this easily with the overlay filesystem option in raspi-config.  (which is apparently using https://github.com/ghollingworth/overlayfs)
Just do:
`sudo raspi-config` then select Performance Options->Overlay Filesystem
Now there should be zero writes to the SD card, and this pi_video_looper still works.

I think this should be added to the readme guide as an optional suggestion.

tofuSCHNITZEL, 2022-09-13:
Thanks for the info - I will include it in the readme!