adafruit/pi_video_looper #44: Suggestion: Safe way to shutdown without SSH

starpause, 2018-04-26:
It would be nice if there was a CTRL-C / Z / \ to exit out of the video looping so the rPi could be safely shut down without SSH access. Thank you!

starpause, 2018-05-04:
@whiteaden agree, that's perfect

@tdicola @subicura @Lyptik anyone around for a zero-conflict PR approval?

ladyada, 2018-05-04:
star, can you test it - let us know it works? :)

starpause, 2018-06-04:
@ladyada I gave the branch for the PR in question a go and it works as expected! Tested with out of the box settings the performance is same as the other rpi I set up with your video looper. 👍

docofjour, 2018-06-20:
I'm kinda new to RPI and to coding stuff as well, thus I'm sorry if my question is a bit lame... :)
We installed 2 RPIs (Model B 1 2012) at an exhibition with hello-video for looping seamlessly. The PIs aren't  easy to access, since they are mounted next to the projectors hanging from ceiling at ca. 4m height. I'm not sure if these older models can handle wireless keyboards just as a plug and play device, but let's assume they can...
My main question is, if it's okay, when I simply copy and paste the green lines [from this commit](https://github.com/adafruit/pi_video_looper/pull/25/commits/18729b8f002ac6630eb997ae2e6701450cafa887) into video_looper.py (and delete the red ones), and do the same with the .ini file? Or can I just replace the original video_looper.py with that one in the commit and edit the .ini as provided?
The exhibition runs until 31th October, and I'm afraid that the installed OS will collapse after a while, since every evening the power is simply switched off in these rooms...
Looking forward for your suggestions, and sorry again for the lame question.

brennen, 2018-07-25:
@docofjour a bit late on this, but I just merged #25, so you should be able to use the current version of `video_looper.py` and the corresponding value in the .ini file.

closed 2018-07-25 by brennen