adafruit/pi_video_looper #50: [help wanted] Composite Output playing severely slower than HDMI output.

whiteaden, 2018-08-13:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):
 **Raspbian OS (tried config edits through both PC & Mac, no change)**

- Python version (run `python -version` or `python3 -version`):  
 **python -version : 2.7.13**
 **python3 -version: 3.5.3**

- Error message you are receiving, including any Python exception traces:
 **I don't receive any error message; the issue I have is that when I output video through composite NTSC (`sdtv_mode=0`) the video plays back at approximately 1/5th of the supposed speed (audio included, so I don't think it's a frame rate issue).

This issue only happens when I boot up the Pi with a composite cable connected *only*, if I am only using the composite output for audio, (HDMI as video output) the composite audio output works at the proper speed (but there is no image being sent).

I've tried this with a few different pi's (the issue occurred after I did a wipe and re-install on all of our 27 pis in rotation)
- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **
 1. install pi_video_looper
 2. verify that sound is sent to **both** Composite & HDMI in video_looper.ini
 2. edit the config.txt file:
  a. disable `hdmi_force_hotplug=1` (comment out or delete)
  b. Add `sdtv_mode=0` to output to composite NTSC
3. connect to composite TV (disconnect HDMI cable, otherwise it'll only output sound through composite)
4. insert USB with compatible video.

the config file including the above edits: [config.txt](https://github.com/adafruit/pi_video_looper/files/2284120/config.txt)

christiansievers, 2019-02-13:
It may have to do with this here:

https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=195178

https://github.com/Hexxeh/rpi-firmware/issues/172

Try adding `audio_pwm_mode=0` to config.txt?

tofuSCHNITZEL, 2019-09-14:
I will close this for now. If this is still relevant please feel free to comment again.

closed 2019-09-14 by tofuSCHNITZEL