adafruit/pi_video_looper #51: Can't exit pi_video_looper with ESC

NormComparini, 2019-03-15:
I have a pi that will be off the network and need to control config settings with a keyboard and mouse. So, I need to be able to exit the looper.

I looked at the many threads and added to one, but we were unable to resolve the issue (https://forums.adafruit.com/viewtopic.php?f=8&t=143884&p=735884#p735884).

Steps so far:
- verified the correct version of the looper, and that it has the ESC code for exiting the program
- verified the .ini file had keyboard control turned on
- checked keyboard and locale settings 
- replaced physical keyboard

Note that I do not see the "press escape" notice.

Any help would be much appreciated!

NormComparini, 2019-03-15:
More data points:

When I edit the ini file to change foreground (text) color, then remove the media drive, I get no change in font color.

When I run the debug on video_looper.py I get this error:
Traceback (most recent call last):
  File "/home/pi/pi_video_looper/Adafruit_Video_Looper/video_looper.py", line 4, in <module>
    import ConfigParser
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 305, in _custom_import
    module = self._original_import(*args, **kw)
ImportError: No module named 'ConfigParser'

tofuSCHNITZEL, 2019-03-15:
Hi,
seems you have used python3 but the original pi video looper is only compatible with python2.
unfortunately this original version seems not to be maintained any more so I forked it and you can
check out my recent update on this project including python3 compatibility and an awesome "copy mode"
I have also included the "ESC" key function but, I have to admit, did not test it yet.

cheers

ladyada, 2019-03-15:
if someone can submit a PR to make it python3 instead of py2 - that would be great :)

tofuSCHNITZEL, 2019-03-15:
@ladyada unfortunately I have no isolated branch with just the py3 changes, because I did all the changes + new and updated features at once
with py3 you also don't need ez_setup anymore
but I will see if I have time to create a py3 branch from the original code

ladyada, 2019-03-16:
thanks tofu! if others in this thread can test the python3 version, that would be great :)

closed 2019-04-15 by tofuSCHNITZEL