adafruit/pi_video_looper #54: Hi has anyone found a way of looping each video for a set amount of time without increasing video length?  eg. A one minute loop cycle for 30mins then proceed to next clip.

magnip, 2019-03-21:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

tofuSCHNITZEL, 2019-03-21:
Hi @magnip,
currently there is no (easy) way to accomplish this, but it seems like an interesting feature. I forked this repository and added a lot of new features!
https://github.com/tofuSCHNITZEL/pi_video_looper

I think your feature request is interesting and I could implement this - what do you think would be the best way to set the playtime for each video?
Maybe via something you add to the filenames (eg. xxxxx_play_30m.mp4 for 30mins or xxxxx_play_30x.mp4 for 30 times - or maybe a playlist file that contains filenames and the definition for how log/often it should play?)

Please let me know your opinion and I will consider programming this feature.

cheers,
Tobias

magnip, 2019-03-21:
> Hi @magnip,
> currently there is no (easy) way to accomplish this, but it seems like an interesting feature. I forked this repository and added a lot of new features! [...]

Hey Tobias thanks for your response. I would prefer a simple file name edit like your eg. That would be ideal. I know if I want say 12 loops 1 every hour to cycle throughout the day I could extend clip length and use a larger usb drive to compensate. But I would like my loops stored in the home folder. With my pi zero looper image at only 1.8gb I could get a full setup plug and play on 4gb cards to switch up in case of corruption or I fancy a change. I will be watching your fork and thanks again for your response.

tofuSCHNITZEL, 2019-03-21:
I would need additional infos from you:
 - implementing the play file x times feature is easy I think
 - implementing the loop file for x minutes is a little bit trickier but also doable - the question is: should the video be forcefully ended if the time is reached or should it play at least till the end after the time limit is reached? (if you want the play until finished behaviour the play x times would solve the problem easier)

or is it something completely different you are after?
could you maybe explain your usage scenario in more detail? thanks.

magnip, 2019-03-21:
Hello again. My scenario is as follows: (what I want works a for single clip only) I have looped a video of 3 baboons preening each other. The clip is around 3 minuets long and takes up around 100mb on usb. Hellovideo loops my clip perfectly so one clip continues to play until I shut the pi down. So far so good... now when you add another video to the usb my 3 minute clip will be 3 minutes and then it will play the next file in que.  Take this cinemagraph for eg. (It’s uploaded as a gif but original is a 7mb video file)
[image]
When I add it to usb brilliant it will loop as long as I want it to, but adding another movie fill force the first to play for only three seconds because that’s real length of footage.
The brilliance of hellovideo is that seemless loop and what it offers to digital art. I don’t care about the transition between videos even though your adding a pic in between is a great addition.
Anyhow long winded I know but let me know if you need me to upload a test clip to gdrive for you to play with. I am researching as well so if I find a workaround I’ll report back here. Many thanks again.

tofuSCHNITZEL, 2019-03-21:
okay I think I get it now. but would it be sufficient for you to set the clip to play 100 times before it jumps to the next or does it have to be time (seconds/minutes...)?

magnip, 2019-03-21:
> okay I think I get it now. but would it be sufficient for you to set the clip to play for example 100 times before it jumps to the next or does it have to be time (seconds/minutes...)?

Yes that’s what I want.  The duration was only to get my understanding across. Playing the same clip over again for a set amount of times before changing to the next clip will be job done!  One thing with route I see arising is the loss of the seemless loop... with hello a single clip is looped with no noticeable effect but a slight skip is seen when moving to next clip. I’m wondering if the clip is ordered to play multiple times will my clip remain seemless.  Either way it’s a new angle if you manage it, and definitely an improvement.👍🏼

magnip, 2019-03-22:
Quick update to hopefully save you time.. I duplicated the same clip on the usb so in theory the same as asking hello video to play same file multiple times and it  showing the 500miliseconds black screen which defeats the object. I need to invoke the skills hello is using when accessing one clip and to keep looping it for a set amount of loops or time..  I have authored my clip to last for 30mins using duplicate features in iMovie which has resulted in around a 6gb file so you can imagine ten clips and I’m already at 60gb. To be honest I’ve been at this for years but got exited when I saw your mods. Cheers for your time though always appreciate the help you have shown.

tofuSCHNITZEL, 2019-03-22:
> it showing the 500miliseconds black screen
ah yes, as I have never used hello_video I never tested if it can seamlessly play multiple files, not only one in a loop

I like your idea and I will implement it anyway. (in fact I'm almost finished) after that I will see if there is a way to get hello_video to play multiple videos seamlessly but I fear this could be difficult

tofuSCHNITZEL, 2019-03-22:
could you maybe provide me with one or two h264 testfiles that should loop seamlessly.

magnip, 2019-03-22:
Hey Tobias here is a drop box link to a shared folder. https://www.dropbox.com/sh/ybqo82smr1pms79/AABaBhnfNT9gTRTjQhXOHigGa?dl=0 Two clips: try each one on its own and experience the seemless loop running endlessly.

tofuSCHNITZEL, 2019-03-22:
hey @magnip,
it's done ;)
check out v1.2.5 of my fork
test it out and tell me if it works the way you need it.
if there is an issue or you have questions please post an issue on my fork.

cheers

closed 2019-04-15 by tofuSCHNITZEL