adafruit/pi_video_looper #55: Installing hello_video: Stuck at unpacking objects 60%

pi43r, 2019-04-26:
- Platform/operating system: Raspbian Stretch -latest#

- Error message you are receiving, including any Python exception traces: 
```
pi@raspberrypi:~/pi_video_looper $ sudo ./install.sh
Installing dependencies...
==========================
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
10 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.3).
git is already the newest version (1:2.11.0-3+deb9u4).
python3 is already the newest version (3.5.3-1).
python3-dev is already the newest version (3.5.3-1).
supervisor is already the newest version (3.3.1-1+deb9u1).
omxplayer is already the newest version (20180905+git061425a-1).
python3-pip is already the newest version (9.0.1-2+rpt2).
python3-pygame is already the newest version (1.9.3+dfsg-2).
0 upgraded, 0 newly installed, 0 to remove and 10 not upgraded.
Installing hello_video...
=========================
Cloning into 'pi_hello_video'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (6/6), done.
Unpacking objects:  60% (25/41) 
```

I don't know why, but the setup seems to be stuck at unpacking pi_hello_video. 
Can anyone confirm the error? What could be the cause?
Any suggestions on how can I fix this?

pi43r, 2019-04-26:
´´´
Cloning into 'pi_hello_video'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (6/6), done.
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: unpack-objects failed
´´´

New error :(

tofuSCHNITZEL, 2019-04-26:
It seems like maybe there is a problem with your sd card or raspbian image...

tofuSCHNITZEL, 2019-04-27:
Please try to flash the latest raspbian (lite) image to your sd card again using https://www.balena.io/etcher/ (which also checks your card for errors after flashing) and try again.

pi43r, 2019-04-28:
I tried to reinstall raspbian, but it still got stuck at 60%. 
Then I used another sd card and it worked. 

I put the backup of the second sd card on the first and it was working, too. Not sure what the problem was ...

Thank you anywway.

closed 2019-04-28 by pi43r

diver-down, 2019-05-01:
Same problem. Unable to solve.