adafruit/pi_video_looper #57: Onboard screen solution? waveshare

quintendewilde, 2019-06-03:
Hello,

How do I get the video looper running with a waveschare screen that is directly mounted on the raspberry pi. Not thru HDMI?

If managed to get the player running more than often on hdmi but never on an onboard screen.

tofuSCHNITZEL, 2019-06-03:
Hi,
maybe one of these could help you?

https://hackaday.com/2018/10/21/blazing-fast-raspberry-pi-display-driver-will-melt-your-face-then-teach-you-how/

https://forum.kodi.tv/showthread.php?tid=330837

Unfortunately I have no SPI screen here at the moment to test it.

Cheers

quintendewilde, 2019-06-05:
Hi, 

Well my screen is mounted on the din rail output but also connected with a small hdmi to hdmi plug. 
I want to have the output to the hdmi. I managed to get the looper working for a few seconds (without following the links you gave me). Then pressed esc and I got back into the Rasbian GUI. 

I tried rebooting and re running the looper but I get back in the GUI. And the moment the GUI loads I get blinking effect to black. I then run the sudo suporvisorctl start video_looper and got 

ERROR: (spawn error) 

Any help would be super handy. And make an otherwise very costless project, very expensive!

tofuSCHNITZEL, 2019-06-06:
> but also connected with a small hdmi to hdmi plug.
> I want to have the output to the hdmi.

so if you have conncted the screen with hdmi it should work like with every other regular display, because the hdmi output is default.

maybe try if it works with a different display first.
if not try to reinstall everything. (including a fresh raspbian-lite image)

please check also the contents of the err log file located in: /var/log/supervisord/

closed 2019-08-24 by tofuSCHNITZEL