adafruit/pi_video_looper #58: Videlooper.ini not found on boot

quintendewilde, 2019-06-19:
I'm installing 30 raspberry pi's and using 2 pi's to install them all. 

1 pi is using a hdmi monitor standard and everything is installing very well.

2 pi is using a 7inch display and the first 3 loopers installed fine. From then on, it says installation finished. But when I try to put the sd card in to my computer to change the directory then the 2th pi doesn't show the videolooper.ini. 
Would be nice to find a solution, because this one installs the fastest.

tofuSCHNITZEL, 2019-06-20:
I'm not sure that I fully understand what you are doing / trying to do. but one solution would be to install a fresh rasbpian lite on one SD card and finish the whole looper setup as usual and then just clone this SD card to the other SD cards (using dd (linux) or drive snapshot (windows))

tofuSCHNITZEL, 2019-07-08:
@quintendewilde any news?

closed 2019-08-24 by tofuSCHNITZEL