adafruit/pi_video_looper #59: Video Looper incompatibilities with adafruit read only filesystem

JesseCake, 2019-07-07:
It seems that there are incompatibilities between the video looper and read only filesystem solutions provided kindly by Adafruit.

To me, these two should be able to go together, as a video looper should be able to have its plug pulled at any point without needing a keyboard etc to shut down.

The video looper appears to run using the systemctl service, however this breaks when the read only filesystem is introduced for some reason. As well as this, once the filesystem is read-only, the mountpoint also cannot be created

Perhaps an addition to the scripts could be the option of having the method of startup happen through a different method, as well as adding to the read only filesystem scripts to have /mnt exist as a tmpfs.

Here are the two things I did to solve the video looper not working after the read only filesystem scripts were implemented (I know they could be more elegant, but they work):

1. Added startup of the video looper to /etc/rc.local with this line added before "exit 0":
`python3 -u -m Adafruit_Video_Looper.video_looper &`

2. Added a mountpoint to /etc/fstab to allow /mnt to exist in ram, and therefore be writeable:
`tmpfs /mnt tmpfs nodev,nosuid 0 0`

I now have a fully functioning read only video looper that can be plugged and unplugged without worry for our installations.

tofuSCHNITZEL, 2019-07-08:
Hi JesseCake,

thanks for sharing your findings.

I researched this topic a while ago and there seems to be no clear answer if a readonly filesystem is really the solution to a "never failing, hard unplug surviving, rpi setup". Because somewhere I read the theory that the sudden loss of power to the SD card controller or SD card can also cause corruption on a "physical" layer.

With this in mind and the troubles I had getting a non buggy readonly mode to work, none of the around 20 players I have "out in the wild" run with a read only file system - some of them get hard powered off on a daily basis and I have yet to see one failing. (of course this can also be just luck)

Maybe the issues with the read only files system comes from the supervisor service that wants to write logfiles?
maybe we dont need your step 1 if we disable logging in "video_looper.conf" by adding
logfile=/dev/null
(maybe also)
nodaemon=true
logfile_maxbytes=0

Would you be able to test this (I'm short of time at the moment)

But I would like to include steps to create a working read only player and a link to the adafruit read only tutorial in the manual here in this repository. And maybe in the future also at learn.adafruit.com.
Or maybe even include a selection in the install script that makes the player read only after the installation.

tofuSCHNITZEL, 2019-08-24:
@JesseCake were you able to check if my suggestions help?

JesseCake, 2019-09-12:
Gosh I'm so sorry, I completely missed this. I'll check in on it and get back to you.

lumbric, 2019-12-12:
This seems to be related to #33.

See also discussion in the [adafruit forum](https://forums.adafruit.com/viewtopic.php?f=8&t=140302).

tofuSCHNITZEL, 2022-02-14:
maybe you can try to use the build from https://videolooper.de/

closed 2022-02-14 by tofuSCHNITZEL