adafruit/pi_video_looper #60: Files on Sd card

Gary202, 2019-08-24:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **INSERT PLATFORM/OPERATING
  SYSTEM HERE**

- Python version (run `python -version` or `python3 -version`):  **INSERT PYTHON
  VERSION HERE**

- Error message you are receiving, including any Python exception traces:  **INSERT
  ERROR MESAGE/EXCEPTION TRACES HERE***

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**

tofuSCHNITZEL, 2019-08-24:
Hi!
you could use supervisorctl restart video_looper over ssh, but you probably would want the player to automatically reload. 
I will have a look if this can be implemented easily!

tofuSCHNITZEL, 2019-08-24:
hey! check out (just released) version 1.0.2.
The directory mode now recognises changes in the number of files in the directory and reloads the playlist.
beware: the file will start playing even if its still copying (e.g. a slow network connection) I've tested this but it seems there were no real issues with it

closed 2019-08-24 by tofuSCHNITZEL

Gary202, 2019-08-24:
one thing when the folder where it pulls the files from is empty and files are added nothing happens till a restart. Whats the command to restart supervisorctl ? I can just do a cron to restart the program everyday?

tofuSCHNITZEL, 2019-08-24:
Are you sure? I will check and fix it if this is the case. You can use the newly added reload.sh to restart the looper or check the bottom of the file for the command you could use directly in a cron.

Gary202, 2019-08-24:
yea I tested it just now but it does update the playlist if a new file is added. I'll just call reload.sh in a Cron job just to make sure it's playing. Thank you for your help. Is there a way to remove the black screen in-between each video?

tofuSCHNITZEL, 2019-08-24:
The pause between each video mostly depends on the speed of your raspberry setup, how fast it can load the files one after another. (if the new delay between video options is 0) 
there is also the possibility to use an image instead of the black background or you could change the background color.

tofuSCHNITZEL, 2019-08-25:
> one thing when the folder where it pulls the files from is empty and files are added nothing happens till a restart.

I just tested it myself with a fresh install and this is not the case. Folder is empty - it displays the no files message; you drop in files (e.g. via smb) it starts playing - I delete everything in the folder it goes back to the no files message

tofuSCHNITZEL, 2019-08-27:
> is it possible to scroll through different pictures in between each video. 
no this is not possible at the moment,

tofuSCHNITZEL, 2019-08-27:
> 
> 
> at the moment but is possible down the road?

I have to think about how to do it - it will probably go together with a playlist feature