adafruit/pi_video_looper #64: aspect mode distortion with Raspberry Pi 4

foocp, 2019-09-09:
Using a Raspberry Pi 4, omxplayer distorts videos to always fill the entire space of the screen. (normal behavior would be to display black bars around the video if the aspect ratio doesn't match)

This happens with versions 1.0.3 and the latest v.1.0.4 of your script. The same SD card when booted on a Raspberry Pi 3b+ doesn't do that.

I wonder if maybe the Pi 4 hardware has problems recognizing the screen size, so I've tried various screens, using both HDMI-to-HDMI cables as well as HDMI-to-DVI cables (when using a desktop screen that only has DVI input). 
These are the results of `tvservice --modes=CEA` and `tvservice --modes=DMT` when connected to a 1920x1200 px screen  - not sure what this means, but maybe it helps?

> pi@raspberrypi:~ $ tvservice --modes=CEA
> Group CEA has 11 modes:
>            mode 1: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
>            mode 2: 720x480 @ 60Hz 4:3, clock:27MHz progressive 
>            mode 3: 720x480 @ 60Hz 16:9, clock:27MHz progressive 
>            mode 4: 1280x720 @ 60Hz 16:9, clock:74MHz progressive 
>            mode 5: 1920x1080 @ 60Hz 16:9, clock:74MHz interlaced 
>   (native) mode 16: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
>            mode 17: 720x576 @ 50Hz 4:3, clock:27MHz progressive 
>            mode 18: 720x576 @ 50Hz 16:9, clock:27MHz progressive 
>            mode 19: 1280x720 @ 50Hz 16:9, clock:74MHz progressive 
>            mode 20: 1920x1080 @ 50Hz 16:9, clock:74MHz interlaced 
>            mode 31: 1920x1080 @ 50Hz 16:9, clock:148MHz progressive 
> pi@raspberrypi:~ $ tvservice --modes=DMT
> Group DMT has 12 modes:
>            mode 4: 640x480 @ 60Hz 4:3, clock:25MHz progressive 
>            mode 8: 800x600 @ 56Hz 4:3, clock:36MHz progressive 
>            mode 9: 800x600 @ 60Hz 4:3, clock:40MHz progressive 
>            mode 16: 1024x768 @ 60Hz 4:3, clock:65MHz progressive 
>            mode 32: 1280x960 @ 60Hz 4:3, clock:108MHz progressive 
>            mode 35: 1280x1024 @ 60Hz 5:4, clock:108MHz progressive 
>            mode 47: 1440x900 @ 60Hz 16:10, clock:106MHz progressive 
>            mode 51: 1600x1200 @ 60Hz 4:3, clock:162MHz progressive 
>            mode 58: 1680x1050 @ 60Hz 16:10, clock:146MHz progressive 
>   (prefer) mode 68: 1920x1200 @ 60Hz 16:10, clock:154MHz progressive 
>            mode 81: 1366x768 @ 60Hz 16:9, clock:85MHz progressive 
>            mode 82: 1920x1080 @ 60Hz 16:9, clock:148MHz progressive 
>

tofuSCHNITZEL, 2019-09-10:
Please try to add
--aspect-mode letterbox
to the additional omxplayer settings at the bottom of the ini file.

foocp, 2019-09-10:
Hi, thanks for looking into this. I tried your suggestion. It makes no difference, it's still stretching the video.

tofuSCHNITZEL, 2019-09-10:
Interesting. Seems like an omxplayer issue to me. So you have a 16:9 video and a 16:10 screen and the video is stretched to fill the 16:10 screen? I will try it today with my 16:10 screen maybe I can replicate the issue.

foocp, 2019-09-10:
Yes, that's exactly what it does. It's the same on an old 1280x1024 (5:4) screen. 
I enabled debug mode in the ini file, but the stdout log doesn't mention anything unusual.

Is there another omxplayer I could try? This here uses a fork of the official one, right?

tofuSCHNITZEL, 2019-09-10:
This project just downloads the one from the raspbian repositories. Have you tried to apt upgrade and maybe there is an updated version?

foocp, 2019-09-10:
Tried that - no. 
I could try compiling from [source](https://github.com/popcornmix/omxplayer); that should be newer, right? Will the video_looper then automatically use the one I compiled?

tofuSCHNITZEL, 2019-09-10:
Video looper just calls "omxplayer" so if you remove the system package and compile it to the bin folder and your build can be called by just using omxplayer in the shell video looper will use your build

foocp, 2019-09-10:
Hmm looks like omxplayer is already completely up to date. Running `sudo apt show omxplayer` gives me `Version: 20190723+gitf543a0d-1` which is the current one from https://github.com/popcornmix/omxplayer.

tofuSCHNITZEL, 2019-09-11:
I just confirmed this issue with my pi4 and 16:10 screen. thanks for opening the issue @ https://github.com/popcornmix/omxplayer/issues/741
lets see what they will figure out

foocp, 2019-09-13:
Fantastic, thanks for finding out! Works for me, too. I also uncommented this for good measure, as it seems to be related to the 2nd screen, which I don't need:
`#max_framebuffers=2`