adafruit/pi_video_looper #65: Playlist feature

Gary202, 2019-09-25:
When will the playlist feature be added for videos and pictures for the load screen??

tofuSCHNITZEL, 2019-09-25:
I'm quite busy at the moment.
what do you mean with pictures for the load screen? You mean a picture at startup or between the videos?

What features do you expect from a playlist which you can't already do by e.g. Naming the files?

Gary202, 2019-09-25:
Yes the picture for the load screen. and play list like play files a,b,c on Monday and play files d,e,f, on Tuesday. I thought this feature was in the works??

tofuSCHNITZEL, 2019-09-25:
No, a schedular is not planned at the moment.
you can grab the script I made for my deployments which tries to hide as much as possible from the rpi boot and also tries to install a loading screen.
you can grab it here:

https://upload.bitconnect.at/files/installer_generic.zip

it will also disable wifi and BT so read through it and maybe remove stuff you don't want or need

closed 2019-09-25 by tofuSCHNITZEL