adafruit/pi_video_looper #70: High CPU Usage

1024jon, 2019-10-22:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspberry pi 3 b+ with Raspbian.  Both version 9.6 stretch and 10 buster

- Python version (run `python -version` or `python3 -version`):  Python 3.5.3 and 3.7.3

- Error message you are receiving, including any Python exception traces:  No errors the software works as intended other than high cpu usage

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): Fresh install of raspbian and pi_video_looper

I have a several of the pi video looper systems running, some of them for years.  I set up a new one and installed the latest version.  I began getting temperature warnings on the pi and noticed that one cpu core was maxed at 100% usage.  I tried different video files, but even without running any video it still uses 100% cpu.  I went to a different pi that has been running the video looper software fine for a few months, updated it to the newest version of video looper and the cpu usage issue followed.  I removed the new version and reinstalled an older version and the cpu issue went away.

tofuSCHNITZEL, 2019-10-22:
What commit IDs do the "old" version without and the "new" version with the problem have?
please post your video_looper.ini

I know that the python code eats a lot of cpu... But I never got more than 30/40 pc (which is still rediculus for the kind of things the py script is doing)

In the future maybe the whole project can be improved by using interrupts instead of having a loop check for usb drive changes etc...

1024jon, 2019-10-22:
Where can I find the commit IDs?

video_looper.ini
# Main configuration file for video looper.
# You can change settings like what video player is used or where to search for
# movie files.  Lines that begin with # are comments that will be ignored.
# Uncomment a line by removing its preceding # character.

# Video looper configuration block follows.
[video_looper]

# What video player will be used to play movies.  Can be either omxplayer or
# hello_video.  omxplayer can play common formats like avi, mov, mp4, etc. and
# with full audio and video, but it has a small ~100ms delay between loops.
# hello_video is a simpler player that doesn't do audio and only plays raw H264
# streams, but loops seemlessly.  The default is omxplayer.
video_player = omxplayer
#video_player = hello_video

# Where to find movie files.  Can be either usb_drive or directory.  When using
# usb_drive any USB stick inserted in to the Pi will be automatically mounted
# and searched for video files (only in the root directory).  Alternatively the
# directory option will search only a specified directory on the SD card for
# movie files.  Note that you change the directory by modifying the setting in
# the [directory] section below.  The default is usb_drive.
#file_reader = usb_drive
file_reader = directory

# The rest of the configuration for video looper below is optional and can be
# ignored.

# Control whether informative messages about the current player state are
# displayed, like the number of movies loaded or if it's waiting to load movies.
# Default is true to display these messages, but can be set to false to disable
# them entirely.
#osd = true
osd = false

# To play random playlist.
is_random = false

# Control the program via keyboard
# If enabled, hit ESC key to quit the program anytime (except countdown).
#keyboard_control = false
keyboard_control = true

# Set the background to a custom image
# This image is displayed between movies
# Can potentially look broken if video resolution is smaller than display resolution
# bgimage = /home/pi/static/bg.png
bgimage =

# Change the color of the background that is displayed behind movies (only works
# with omxplayer).  Provide 3 numeric values from 0 to 255 separated by a commma
# for the red, green, and blue color value.  Default is 0, 0, 0 or black.
bgcolor = 0, 0, 0

# Change the color of the foreground text that is displayed with the on screen
# display messages.  Provide 3 numeric values in the same format as bgcolor
# above.  Default is 255, 255, 255 or white.
fgcolor = 255, 255, 255

# Output program state to standard output if true.
console_output = true
#console_output = false

# Directory file reader configuration follows.
[directory]

# The path to search for movies when using the directory file reader.
path = /home/pi/Videos

# USB drive file reader configuration follows.
[usb_drive]

# The path to mount new USB drives.  A number will be appended to the path for
# each mounted drive (i.e. /mnt/usbdrive0, /mnt/usbdrive1, etc.).
mount_path = /mnt/usbdrive

# Whether to mount the USB drives as readonly (true) or writable (false).  It is
# recommended to mount USB drives readonly for reliability.
readonly = true

# omxplayer configuration follows.
[omxplayer]

# List of supported file extensions.  Must be comma separated and should not
# include the dot at the start of the extension.
extensions = avi, mov, mkv, mp4, m4v

# Sound output for omxplayer, either hdmi, local, or both.  When set to hdmi the
# video sound will be played on the HDMI output, and when set to local the sound
# will be played on the analog audio output.  A value of both will play sound on
# both HDMI and the analog output.  The both value is the default.
sound = both
#sound = hdmi
#sound = local

# Sound volume output for the video player will be read from  a  file  near  the
# video files. If the file does not exist, a default volume of 0db will be used.
# To use this feature create a file in the same directory as the videos and name
# it the value defined below (like 'sound_volume' by default), then inside the
# file add a single line with the volume value in text to pass to omxplayer (using
# its --vol option which takes a value in millibels).
sound_vol_file = sound_volume

# Any extra command line arguments to pass to omxplayer.  It is not recommended
# that you change this unless you have a specific need to do so!  The audio and
# video FIFO buffers are kept low to reduce clipping ends of movie at loop.
extra_args = --no-osd --audio_fifo 0.01 --video_fifo 0.01

# hello_video player configuration follows.
[hello_video]

# List of supported file extensions.  Must be comma separated and should not
# include the dot at the start of the extension.
extensions = h264

tofuSCHNITZEL, 2019-10-22:
Since you use directory mode maybe the newly introduced check for changes in the directory is causing the high cpu for you, or maybe the keyboard shortcut checks...
could you please disable the KB shortcuts or try the USB drive mode and see how the cpu usage changes?

Also I recommend to disable the console output as it can fill your SD card over time

1024jon, 2019-10-22:
Seems like disabling keyboard control fixes it.
[image]
[image]

closed 2019-10-22 by 1024jon