adafruit/pi_video_looper #73: "Insert USB drive with compatible movies" not shown on rotated Screen

glepping, 2019-10-29:
I have set up a Pi 4 (1GB) with parameter display_hdmi_rotate=1 which rotates the screen 90 degrees clockwise. With these settings the "Insert USB drive with compatible movies" sentence is not shown in idle mode.

Tried to shorten the phrase and reinstalled video looper again without success.

Any suggestions?

tofuSCHNITZEL, 2019-10-29:
interesting... osd = true in ini? videos etc work?
as soon as my rpi 4 is back I will test this and maybe also with a rpi 3

glepping, 2019-10-30:
it is working on a Pi 3B+ (I used the same microSD with Pi 4)

tofuSCHNITZEL, 2019-11-21:
I played around a bit with the pi4 and there seems to be quite a lot of change regarding the output becasue display_hdmi_rotate=1 does not always work and it depends on which video driver is used. I only managed to rotate the screen with display_hdmi_rotate=1 once and it also did not play video in this configuration but after installing raspbian again I did not manage to rotate the screen again. 
but there is probably a problem with the rotating and the driver and how pygame outputs to the framebuffer... but unfortunately I don't know enough about the new rpi4 display drivers to debug further