adafruit/pi_video_looper #74: Keyboard input is disabled on Raspberry Pi 4

dancind, 2019-10-30:
Hello, I posted on [forums.adafruit.com](https://forums.adafruit.com/viewtopic.php?f=50&t=158018) and was told to inquire here, so here I am. I will post the same thing I wrote there.

Hello, this program has been working great for me and doing exactly what I wanted, but now that I have upgraded to a raspberry pi 4, I can no longer cancel video once playback has started. I know that it is a problem with the 4 because I can still cancel playback using the escape key when I put the sd card into a raspberry pi 3. Please let me know, I really enjoy having this capability.

tofuSCHNITZEL, 2019-10-30:
okay there seems to be a lot of weird stuff with the rpi4. I hope someone with more experience rpi4 hardware related can help out.

NotThatDrunk, 2020-01-29:
I've installed on an rpi4 and have the same issue. I have full control remotely over ssh to stop/start/restart the player through the "sudo supervisorctl" commands. Keyboard commands in the player are not working, even while simulating keyboard presses using various methods through terminal (xdotool).

Seems like "supervisorctl" is a command to kill/stop processes, but is there a way to create new supervisorctl commands such as skip and pause? I can hotkey common terminal commands to my "remote" (phone ssh) if it were to work.

tofuSCHNITZEL, 2022-02-14:
Update: last time I did some testing on the RPI4 it seems kb shortcuts were working fine... does it still not work for you? which OS are you using?

tofuSCHNITZEL, 2023-11-16:
KB shortcuts are working fine on RPi4 - closing

closed 2023-11-16 by tofuSCHNITZEL