adafruit/pi_video_looper #76: Looper does not function on forced reboot

popeyepie, 2019-11-06:
Hi there. 
The Adafruit Looper is great. Unfrotunately, when tehre is a powercut, raspbian is reebooting but then the screen stays black when the video from the usb should start and then loop. I first have to reboot WITHOUT the usb-stick+video and then the looper "asks" to insert the usb stick again. 
Any idea ? 
Thank you

tofuSCHNITZEL, 2019-11-06:
never had this issue before. will test if this is really the case.

popeyepie, 2019-11-06:
thank you. 
i’ve tried everything. tonight i formatted the usb drive (ntfs) via raspberry and not on my mac, still the same problem. in fact, when I stop the loop via ESC, raspberry wants me to put my password so that it is mounted or sth like that.
i don’t use raspbian lite, i use raspbian full/standard. 
when i put the files on the sd card, means not on USB-DRIVE but on DIRECTORY, the loop was there even if there was a forced reboot (e.g. no electricity).
but when i tried to use the COPYFILES mode so that the latest version of a video is copied and rewritten on the sd card, it doesn’t function again as ada fruit or raspbian kind of can’t access the drive. 
it only accesses the drive when the window opens “please insert usb drive to play video”. 
does my info help ?
thank you for your help.
kind regards,
marin

tofuSCHNITZEL, 2019-11-06:
I just installed a fresh raspbian Lite on an SD card and used it in a RPI3B.

in usb mode, leaving the usb plugged in, (its formatted FAT32 in my case)
the player starts playing regardless of hard reboot or soft reboot

I cannot stress this enough: use raspbian lite.
Because the regular raspbian window manager / GUI might interfere with usb mounting etc.

you can also check the supervisor logs in:
/var/log/supervisor/video_looper-std*

closed 2019-11-06 by tofuSCHNITZEL