adafruit/pi_video_looper #77: Switching from omxplayer to hello_video - MALFUNCTION

popeyepie, 2019-11-06:
hi there. 
In tha adafruit descriptiomn it is written, that the configuration of the looper may be done easily in an texteditor. i've chhanged from 
video_player = omxplayer
#video_player = hello_video 
to
#video_player = omxplayer
video_player = hello_video
but then on reboot of raspbian, the screen stays on rasbian and no looper starts (even if file is on raw .h264).
i do it with an apple mac.
i had to reinstall the looper so it functions. 
Any idea ?
Thank you

tofuSCHNITZEL, 2019-11-06:
HI,
never had this issue before and I always test the installation in every possible combination.
after a config change you can use reload.sh or rebooting the pi as well.
what raspian image did you use?
I strongly suggest you use the latest raspbian LITE image and use https://www.balena.io/etcher/ to copy and verify the sd card.

tofuSCHNITZEL, 2019-11-06:
I just installed a fresh raspbian Lite on an SD card and used it in a RPI3B.
changing the config and running ./reload.sh works fine.
must be a broken install on your end.

I do not recommend using the full raspbian - the usage of lite is the way to go since this software is designed to run "stand alone" on an rpi.

but it never hurts to look in the log files maybe you will see whats is causing your problems:
cat /var/log/supervisor/video_looper-std*

closed 2019-11-06 by tofuSCHNITZEL