adafruit/pi_video_looper #79: Playlist file

Gary202, 2019-11-06:
What does a example play list file look like????

tofuSCHNITZEL, 2019-11-06:
I have added an example m3u file. an easy way is also to use e.g. VLC (but beware that the filepaths may be absolute)
also the internet has good explainations and examples:
https://en.wikipedia.org/wiki/M3U

closed 2019-11-06 by tofuSCHNITZEL

Gary202, 2019-11-06:
does the play list feature work with hello_video??

tofuSCHNITZEL, 2019-11-06:
as it is only responsible for the generation of the playlist it is independend of the player. but the title display only works with omxplayer

Gary202, 2019-11-06:
ok thank you

Gary202, 2019-11-06:
maybe a dumb question is there away to update a instance all ready installed??

tofuSCHNITZEL, 2019-11-06:
remove the existing pi_video_looper folder. 
clone this repository again. `git clone https://github.com/adafruit/pi_video_looper/` via ssh
then just run `sudo ./install.sh `
(inside the pi_video_looper)

Gary202, 2019-11-06:
ok

Gary202, 2019-11-06:
question would this work
#path = *.m3u
I would like to use any playlist in the video directory? I'm updating the directory with a script "only filename changes are downloaded. Would like to use the date for the name of the playlist (11-6-19.m3u). So i can update the playlist and the script would download the changes?

tofuSCHNITZEL, 2019-11-06:
unfortunately only a fixed filename is supported