adafruit/pi_video_looper #80: Streaming Support

sir-rave-alot, 2019-11-12:
Hi there,
I'd like to extend the current code for supporting the following usecase:
instead of an attached display, the video content should be feed to a webserver (apache , ffmpeg stream ...) instantly, so "streaming" the videos from any host in the network using a  webbrowser is possible.

I like the robustness of the mount/remount ... etc routines, so building upon this would be nice.

Any suggestions, where I can start?
Is it possible to just ''substitute" the omxplayer with apache?

It would be helpful to point out, which files depend on the "video player".

Best regards.

christiansievers, 2019-11-20:
While a streaming server like you describe would clearly have a lot of use cases, can we please make sure not to slowly kill pi_ video_looper (a great project) will feature bloat? 

It has been rock stable for many years now, and only recently seems to have become somewhat unpredictable and complicated with the introduction of the Raspberry 4. Now what (I at least) would really like to see is to make it simple again. Not expand it even more. 

@sir-rave-alot I guess a way forward would be to fork this repository, use what works for your purposes and get rid of the bits you cannot use, i.e. local playback.

tofuSCHNITZEL, 2019-11-20:
@christiansievers yes you are absolutely right. after adding the copymode (which I think is a very usefull addition - I use it on all projects) I had the feeling of feature creep ;)
For me the future of pi_video_looper lies in optimisations and making it work with the RPi4

closed 2019-11-20 by tofuSCHNITZEL