adafruit/pi_video_looper #81: Home folder on pi

Gary202, 2019-11-20:
I have pi user name changed for security resons. I see when installing the pi_video_player it creates "/home/pi/video" were could i cange the install script to use the currect home folder "$HOME/video"  when installing??

Gary202, 2019-11-20:
or on the next version could the changed be included??

tofuSCHNITZEL, 2019-11-20:
Sure good point. I will try to do it today.

tofuSCHNITZEL, 2019-11-20:
the creation of the folder in /home/pi/video with the installer is just for convenience. just create a folder yourself (wherever you want) and enter the path in /boot/video_looper.ini

closed 2019-11-20 by tofuSCHNITZEL

Gary202, 2019-11-21:
Correct but since I'm not useing the pi user and the home folder is a different name. Once the installation is completed I have to remove the pi/ folder the installation creates. If your installation uses $HOME it will create a folder in the current home directory