adafruit/pi_video_looper #83: Video freezing

Gary202, 2019-12-03:
What would cause video freezing?? The os is not frozen? just the video
My set up
Board = Pi zero w
OS = Rasbian stretch lite
Player = Hello video
ram split = 128
sd card = Sandisk 16 gig Class-10 UHS-1 A1

tofuSCHNITZEL, 2019-12-03:
Please post your video_looper.ini

Gary202, 2019-12-10:
Sorry for the let reponse was on vacation
The video freezes randomly  here is the video_looper.ini
[video_looper.ini.txt](https://github.com/adafruit/pi_video_looper/files/3944239/video_looper.ini.txt)
and i have around 12 or 13 .h264 files

tofuSCHNITZEL, 2019-12-10:
please try to disable keyboard_control.
also maybe try with a differnet SD card or maybe the files are somehow corrupted and hello cannot handle them...
is there a reason why you would use hello video? why not use regular mp4 h264 files and use omxplayer?

Gary202, 2019-12-10:
I need the seemless playback of several files back to back omxplays has a pause in between each file.

tofuSCHNITZEL, 2019-12-10:
as far is I know there will be a gap between different files even with hello_video

Gary202, 2019-12-13:
The gap with omxplayer on the pi zero is around 3. seconds the gap with hello video is around .8 seconds

tofuSCHNITZEL, 2019-12-13:
Have you tried to disable the keyboard control?
Have you tried switching the ad card?

Gary202, 2019-12-20:
yes i have tryed and the video just hangs not sure what the issue is

tofuSCHNITZEL, 2019-12-20:
Maybe the issue somehow is with the video file itself or with the rpi hardware.
Would it be possible for you to send me the video and I try it?

Gary202, 2019-12-24:
so with the keyboard control disabled i see the cpu load drop 50%!! on the pi zero with keyboard controller enabled the cpu load is around 85% with it disabled the load is around 35% is this correct? with the keyboard control disables how do i quiet the video playback???

tofuSCHNITZEL, 2019-12-24:
Over ssh would be the only possibility at the moment.

Gary202, 2019-12-27:
video freezing has stoped not sure what the issue was??
I have disabled the keyboard controlls

closed 2019-12-27 by Gary202

tofuSCHNITZEL, 2020-02-02:
> 
> 
> video freezing has stoped not sure what the issue was??
> I have disabled the keyboard controlls

if you have time maybe try it with the v1.0.7 - the cpu usage caused by the keyboard control handling was improved there