adafruit/pi_video_looper #84: Stop script over VNC

adinhalilovic, 2019-12-10:
Hi,
i'm using raspberry pi 3 with raspberry desktop raspbian. I need to have control of my raspberry over VNC, since its far away from me. When i connect to raspberry with VNC i cant stop video(ESC dont work), actually cant do nothing since my screen is all black. Is it possible to help me fix this ? Can i fix something in source ?

christiansievers, 2019-12-10:
Can’t you control it via ssh? That works like a charm.

adinhalilovic, 2019-12-11:
I can't do port forwarding since i don't have access to router

tofuSCHNITZEL, 2019-12-11:
but VNC works?
you could also try https://www.howtoforge.com/reverse-ssh-tunneling

adinhalilovic, 2019-12-11:
Yea, VNC works fine, I can connect to my rasp, but i only got a black screen and cant do anything.

christiansievers, 2019-12-11:
and port forwarding for VNC is set up on the router?

adinhalilovic, 2019-12-11:
I installed vnc server on raps and just log in into my account. On other side(my laptop) i installed vnc viewer app and thats it. There is no need to change anything about ports and stuff.

christiansievers, 2019-12-11:
- when using VNC, are you on the same wifi or LAN or accessing the Raspberry from the outside?
- if you're local (and not from remote), logging in via SSH will work easily. Enable it from the Raspi's preferences, or in the Terminal via "sudo raspi config" first. Find out the Raspi's IP and log in from your computer with "ssh pi@(ip-address-here)", i.e. "ssh pi@192.168.0.2"
- once logged in, "sudo supervisorctl stop video_looper" will stop it. Likewise, "sudo supervisorctl start video_looper" will start it. Use the same with "restart", to, well you get it. 

- if you're logging in via VNC from remote, then there's already some kind of port forwarding set up. Find out what is set up and use that to also set up port forwarding for ssh. Be careful with these things, as you're basically inviting the whole big bad Internet into your local network. Make as the absolute minimum sure to give the Raspi a good and strong password.

tofuSCHNITZEL, 2023-11-16:
Stale

closed 2023-11-16 by tofuSCHNITZEL