adafruit/pi_video_looper #86: Enable CopyMode

euripide2913, 2020-01-13:
Hi there, sound really silly but could you please guide me through on how to enable the CopyMode Function? I've been trying to enable this function to copy the video file on the pi but it's not working :(

tofuSCHNITZEL, 2020-01-13:
first you have to enable the copymode via the ini.
set `file_reader = usb_drive_copymode` in /boot/videolooper.ini

take note, or better change the password in the "copymode" section of the ini file.

then you fill a usb drive with the videofiles you want on the player and then you create a file that has the same filename as the password set earlier (a file without file extension)

if you plugin this drive to the player you should see the copy progress window

euripide2913, 2020-01-13:
Hi Tobias,
Thank you buddy for replying, I have a better idea now on how to do it!
Really appreciate your help on this
Cheers,
jp

closed 2020-02-02 by tofuSCHNITZEL