adafruit/pi_video_looper #89: Wearing out SD card

philggg, 2020-04-09:
I use the video looper to show a video in a fake fireplace of a video. Unfortunately, it keep wearing out the SD card (not the USB stick with the video). I have tried 3 different types Pi, different versions of Raspian. I have followed typical recommendations to fix flash wear like: removed logging from the OS, removed the swapfile and yet within a few days most of the flash memory in the SD card is gone. I have used large and small video files and the same thing happens. I can only surmise that for some reason the video is constantly being written and read from the SD card and that is the cause of the flash wear out. The SD cards are all high quality SanDisk card with 32 GB of RAM. The video's are less than 50 MB. 

Any suggestions on how to fix this??
- List the steps to reproduce the problem below (if possible attach code or commands
  to run): **LIST REPRO STEPS BELOW**
Run a video for at least a week.  Check flash memory size when you start.  Recheck flash memory size after a week.  (Memory available on flash card will be substantially reduced)

tofuSCHNITZEL, 2020-04-10:
>  most of the flash memory in the SD card is gone. 
so this does not mean the flash is worn out but its full? 

if console_output = false is set in the ini video_looper shoud not generate anything in /var/log/supervisord/....

philggg, 2020-04-10:
The way you know the Flash is wearing out is that the flash memory (on the SD card) starts off with 32 GB and then over days shrinks until it is just 50 MB.

ocouch, 2021-05-02:
@philggg Are you sure you're not confusing available space with disk size?

SD cards don't shrink this much - the internal controllers can't handle it.

Try re-partitioning and reformatting the card - if you use the Raspberry Pi imager's reformat option, it'll whipe the partition table to a single partition as well as format it to FAT32. Give this a go and see if your cards are still ending up with 50Mb remaining.

https://www.raspberrypi.org/software/

philggg, 2021-05-02:
Thanks for taking the time to comment, but my observations were correct.  I did more experimenting (including reformatting) and the SD cards are trashed. (I checked reformatting them on a MAC and Windows machine also)  After more work, I learned that if you have a large video (larger than available free RAM) that it causes paging, and that it will eventually trash the flash memory. (Actually it will get trashed quite fast, in less than a month). 
The work around is to only use very small video files, (must be much less than available RAM), so that no paging occurs when running.

burnburnrocket, 2022-07-05:
Saw this issue was still open; is this still an issue on current releases of video looper?

tofuSCHNITZEL, 2022-07-05:
I don't think so. Using a high quality sd card I have video loppers running for months without problems

closed 2022-09-13 by tofuSCHNITZEL