adafruit/pi_video_looper #93: Scaling to smaller screens.

Lt-Lemming, 2020-06-13:
I'm using Pi Video Looper with Pi Zero's running some 5" screens for a project.

These screens are only 800 x 480 resolution and are configured as such in the Pi's config.txt

One of the issues I have run into with it is that the error messages and the like don't scale to the smaller screen resolution. So I quite often end up with text that is off the screen and unreadable.

tofuSCHNITZEL, 2020-06-14:
Hi,
thanks for the feedback. could you try the version in testing for me?
https://github.com/adafruit/pi_video_looper/tree/testing

`git clone --single-branch --branch testing https://github.com/adafruit/pi_video_looper/`

it should scale down the text for low-res screens - but I'm not sure if its enough.

Thanks!

sirib123, 2021-12-01:
Hi! Was there any update on this? I've got a similar issue (using a 480x480 round hyperpixel screen). I've got letter boxing top and bottom only, and when I input a video file that is 480x480, it keeps the black letterbox on the top and bottom, but also stretched the video horizontally. I'd be incredibly grateful for help with this, as I'm needing it to work full screen for a deadline!

tofuSCHNITZEL, 2021-12-01:
the above changes only affect the text size - it changes nothing with the output resolution...
do you have the bars also on bootup? do you have bars if you use a rpi with an os that has a desktop? how is the screen connected? if its HDMI you can try to change the overscan option for the rpi? (https://raspberrypi.stackexchange.com/questions/23094/calibrate-overscan-settings)

sirib123, 2021-12-01:
Hi! Thanks for getting back so quickly! I've connected the Screen via the GPIO pins so it sits like a HAT on the pi. Booting up and the Raspberry pi desktop all look fine, no stretch, and no black borders. It's only once I launch videolooper that the text is cut off at each side ('insert USB') almost like it was forced into portrait mode. Then when I insert the file it stretches it, and the black bars remain on the left and right.

tofuSCHNITZEL, 2021-12-01:
hm maybe its an issue with how omxplayer outputs the raw video data to the system.. unfortunately I have no experience with non hdmi screens on the pi... maybe you should also try a raspi os lite without desktop

sirib123, 2021-12-01:
Yes I think it must be omx-player related, although I did try using the hello-video option too, and had the same issues! Are there any keywords you'd suggest searching?

tofuSCHNITZEL, 2021-12-01:
Could you give me some more infos:
What make/model is the screen you are using?
Could you maybe provide a picture of how it looks:
On desktop
With the text displaying
With a video in 480x480 resolution paying
With a video in 1920x1080 resolution playing

sirib123, 2021-12-02:
Hi! the screen is the 2.1 round screen from Hyperpixel (https://shop.pimoroni.com/products/hyperpixel-round)
I can take another picture of the desktop tomorrow, but it's centred and fills the screen, I can even see part of the tool bar at the top. Here's a test file I played via USB on the video looper;
Here's the original 16:9 file
[image]
Here it is on the round screen (squeezed and letterboxed)
[image]
Here's the same file, but cropped to 480x480px
[image]
and here is 480x480 on round screen, also squeezed and letterboxed, but this time letterboxed on left and right, not top and bottom like the 16:9 one (sorry for low quality!):
[image]
This is the same for a 1080x1080 file:
[image]
Very very grateful for any help with this and thanks for taking the time!!!

kiswida, 2021-12-05:
hello, first of all sorry about my bad english. I have a Raspberry Pi Zero v.1.3, 1 GHz, 512 MB RAM and a 5 "display with hdmi and 800 * 480 resolution. Unfortunately, they don't work together. the pi zero whit a 27" fullhd monitor works fine, and the display whit my pc hdmi out is even fine together just gray strips... i took some pictures.
please help
than
[image]
[image]
[image]
ks

tofuSCHNITZEL, 2021-12-06:
@sirib123 
have you tried the different aspect mode options for omxplayer?

--aspect-mode

Letterbox, fill, stretch. Default: stretch if win is specified, letterbox otherwise

you can set this in the videolooper ini

tofuSCHNITZEL, 2021-12-06:
> does the screen work if you install a full rasperrypi os with desktop? I'm not sure this kind of thing can be caused by the looper programm... (do you see the pi bootup text?)

kiswida, 2021-12-07:
Hy, No it doesn't work, it just shows these gray sh*t all the time.

Thanks

sirib123, 2021-12-09:
> @sirib123 have you tried the different aspect mode options for omxplayer?
> 
> --aspect-mode
> 
> Letterbox, fill, stretch. Default: stretch if win is specified, letterbox otherwise
> 
> you can set this in the videolooper ini

I believe I did try one of these (Fill?) and it came out with the same result. I don't have access to the screen at the moment, but will get back to you when I get access to it again and give it a go! (For now, as a temporary fix, I have stretched the video I was wanting to play in Premiere Pro in the opposite direction, and so it comes out un-stretched on the screen, albeit with a black border)

kiswida, 2021-12-09:
hello,

i find the solution on youtube: https://www.youtube.com/watch?v=TYzOhgUmZcw&list=LL&index=3&t=42s

1.  create a ssh file
2. edit the config.txt: 
3. 
max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt=800 480 60 6 0 0 0                # 800x480 on my display....
hdmi_ignore_edid=0xa5000080
config_hdmi_boost=2

Thanks !

tofuSCHNITZEL, 2021-12-09:
great find!

jayhoenes, 2023-04-14:
@sirib123 
I got the same problem with the hyperpixel screen and the pi_video_looper.
I think it's out of date now, but maybe it will help others with the same problem.

@tofuSCHNITZEL was right with the --aspect-mode, but it was hard for me to find the right place where to put the code. So here is the solution:

in /boot directory there is a video_looper.ini

Look for the line extra_args = .....  
here you add --aspect-mode fill
like this:

extra_args = --no-osd --audio_fifo 0.01 --video_fifo 0.01 --align center --font-size 55 --aspect-mode fill

tofuSCHNITZEL, 2023-04-14:
thanks for pointing that out. sorry if I wasn't clear enough.

closed 2023-04-14 by tofuSCHNITZEL