adafruit/pi_video_looper #94: Keydown video triggers

elampox, 2020-06-24:
I was trying to find a way to add key binds to trigger specific videos in a path
"Ex: 0->(play video path 0), 1->(play video path 1), 2->(play video path 2)" 
And at the end of the videos return to the main idle loop.

The usage case is that you have a screen showing your idle videos in a loop, but at keypress (numeric 0-9) you trigger a specific video and go back to idle loop.

I got to the part of editing  "def _handle_keyboard_shortcuts(self):" to add the KEYDOWN events and load some video paths from the config file with config.get() , but couldn't figure our how to create a Movie class instance and send it to the Player class without breaking something.

tofuSCHNITZEL, 2020-06-25:
Hi,
I don't thinks thats so easy to acheive... since everything is handled in run() you have to somehow load a specific movie from the playlist and avoid that just the next is played...
I think it needs a rework of how the whole thing works to get this to work. though I think this would be a nice feature...

tofuSCHNITZEL, 2023-11-16:
this is now possible:
e.g.:
```
if event.key == pygame.K_KP5:
    self._playlist.set_next("test.mp4")
    self._player.stop(3)
```

closed 2023-11-16 by tofuSCHNITZEL