adafruit/pi_video_looper #95: Change hotkeys shortcuts

jfou87, 2020-07-03:
Hello ! And thank you for the beautiful work :-)

I liked so much this simple, but powerful code. 
I had just a small question. 

I try to control the video looper via a USB Keypad in place of a regular keyboard (place saver)
I tried to change the k and s hotkey in the videolooper.py like this by using 1 in place of k and 2 in place of s:

def _handle_keyboard_shortcuts(self):
        while self._running:
            event = pygame.event.wait()
            if event.type == pygame.KEYDOWN:
                # If pressed key is ESC quit program
                if event.key == pygame.K_ESCAPE:
                    self._print("ESC was pressed. quitting...")
                    self.quit()
                if event.key == pygame.K_1:
                    self._print("1 was pressed. skipping...")
                    self._player.stop(3)
                if event.key == pygame.K_2:
                    if self._playbackStopped:
                        self._print("2 was pressed. starting...")
                        self._playbackStopped = False
                    else:
                        self._print("2 was pressed. stopping...")
                        self._playbackStopped = True
                        self._player.stop(3)

But the oldest hotkeys still working, and not the new. 

Can you tell me to fix it ?

tofuSCHNITZEL, 2020-07-03:
Please paste your code with the "code" formatting otherwise there are no indents and I can't read it properly.

jfou87, 2020-07-03:
`def _handle_keyboard_shortcuts(self):
        while self._running:
            event = pygame.event.wait()
            if event.type == pygame.KEYDOWN:
                # If pressed key is ESC quit program
                if event.key == pygame.K_ESCAPE:
                    self._print("ESC was pressed. quitting...")
                    self.quit()
                if event.key == pygame.K_1:
                    self._print("1 was pressed. skipping...")
                    self._player.stop(3)
                if event.key == pygame.K_2:
                    if self._playbackStopped:
                        self._print("2 was pressed. starting...")
                        self._playbackStopped = False
                    else:
                        self._print("2 was pressed. stopping...")
                        self._playbackStopped = True
                        self._player.stop(3)`

Sorry, is it better ?

jfou87, 2020-07-03:
```def _handle_keyboard_shortcuts(self):
        while self._running:
            event = pygame.event.wait()
            if event.type == pygame.KEYDOWN:
                # If pressed key is ESC quit program
                if event.key == pygame.K_ESCAPE:
                    self._print("ESC was pressed. quitting...")
                    self.quit()
                if event.key == pygame.K_1:
                    self._print("1 was pressed. skipping...")
                    self._player.stop(3)
                if event.key == pygame.K_2:
                    if self._playbackStopped:
                        self._print("2 was pressed. starting...")
                        self._playbackStopped = False
                    else:
                        self._print("2 was pressed. stopping...")
                        self._playbackStopped = True
                        self._player.stop(3)```

jfou87, 2020-07-03:
I think it's better. 
Sorry, I'm a newbie in the issue section of GitHub

tofuSCHNITZEL, 2020-07-03:
Do you want to control it with the numberpad?
If so the constants for the numberpad keys are:

`K_KP0                 keypad 0
K_KP1                 keypad 1
K_KP2                 keypad 2
K_KP3                 keypad 3
K_KP4                 keypad 4
K_KP5                 keypad 5
K_KP6                 keypad 6
K_KP7                 keypad 7
K_KP8                 keypad 8
K_KP9                 keypad 9`

jfou87, 2020-07-04:
Thank you for your help. 
Am I right if I modify this part of the code in the video looper.py ? 
Or is there somewhere else to change ?

Thank you for your answer

tofuSCHNITZEL, 2020-07-04:
Yes you only need to change this to change the key bindings.

jfou87, 2020-07-04:
It's doesn't work :'(

tofuSCHNITZEL, 2020-07-05:
You have to rerun the installer after you change the python code so that the module gets updated.

`python3 setup.py install --force`

tofuSCHNITZEL, 2022-02-06:
If you still have issues please open this again.

closed 2022-02-06 by tofuSCHNITZEL