adafruit/pi_video_looper #96: Newer version of Raspian make stuttery video in AV output,

TonyTowersTurkey, 2020-07-19:
Hello, first of all, thank you very much for this code, this program has allowed me to substitute old DVD players in a Hotel in PR to service TV channels for the rooms.  your code has been very helpful. 

I want to report an issue I have noticed with the video looper.

in my usage, I use the video to play local movies into a HeadEnd server that only accepts analog input, so I use the Raspberries AV output. I have noticed that in the newer versions of the Raspian starting from Strech the output of the video is very stuttery, to the point that is unwatchable. This is still true in the most recent version to date of Raspberry OS 2020-05-27

I am stuck using an old version of raspbian-2017-07-05  (does not run on the most recent versions of Raspberry pies and I'm concerned about the future. 

Again your project has been a great help for my business and would like to donate to your project.

TonyTowersTurkey, 2020-07-19:
Turns out is an issue with OXMplayer. 
in the config file if you change the audio output to AV instead of auto it fixes the issue. 

this is the forum that helped me
https://retropie.org.uk/forum/topic/17107/bug-solved-omxplayer-on-stretch-based-builds-stutters-on-composite/53?_=1595189973663&lang=en-US

tofuSCHNITZEL, 2020-07-20:
Hi, thanks for your findings.. could you please summarise here all changes you had to make to get it to work so I can add it to the readme?

TonyTowersTurkey, 2020-07-20:
Clarification this is for using the AV (component) output and no HDMI

Very simple,
The only thing to change is in the config `video_looper.ini` 

comment line 169 `#sound = both`
and uncomment line 171 `sound = local`

this forces OXMplayer to use the AV (composite) audio output and fixes the stutter issue. 
the HDMI output still works but there is no audio output because is forced to the AV

closed 2020-07-20 by TonyTowersTurkey