adafruit/pi_video_looper #98: Uninstallation instructions / uninstall script

hellocatfood, 2020-08-28:
Are there instructions for uninstalling the video looper or is there an uninstallation script? I've seen there is a disable script but I'm looking to uninstall it (don't want reinstall the OS and start again)

tofuSCHNITZEL, 2020-09-04:
Hi,
I would say disable and remove all the files will be very close to "uninstalling".
in addition you could apt remove the packages that were installed (see install.sh)
not really sure on how to remove the python package but its not really doing anything if it just stays there...
cheers

yeyeto2788, 2020-09-23:
Hey! I just came up to this issue and I hope this helps.

1. To uninstall python libraries you can do `pip uninstall <package_name>` the only package being install is `setuptools` which is actually useful if you're planning to do more things in python

2. You can also delete `/boot/video_looper.ini` file which is being copied on line https://github.com/adafruit/pi_video_looper/blob/master/install.sh#L46. And also `/etc/supervisor/conf.d/video_looper.conf` copied as in https://github.com/adafruit/pi_video_looper/blob/master/install.sh#L46

3. The apt packages installed by the `install.sh` script are `python3 python3-pip python3-pygame supervisor omxplayer ntfs-3g exfat-fuse` and depending on the arguments passed to the script `git build-essential python3-dev` could be also installed. More info > https://github.com/adafruit/pi_video_looper/blob/master/install.sh#L15 and https://github.com/adafruit/pi_video_looper/blob/master/install.sh#L21

4. Delete created directories `/mnt/usbdrive0` and `/home/pi/video`

5. Delete cloned repository.

closed 2021-12-01 by tofuSCHNITZEL

Chrisgraue, 2023-12-05:
You'll have to forgive me, I'm a Windows user with limited Pi experience. I'd like to update to get the new features.

I tried running the pip uninstall script, but it says that's a syntax error.

I see none of the file directories mentioned when I put the SD card in my laptop to try and make edits. There are a bunch of various txt, ini, and dtb files. The only folders are:

fseventsd
spotlight-v100
TemporaryItems
overlays

None of these contain a mnt, home, or pi subdirectory. The only thing I see (as a dumb person) that references the video looper at all is the pi_video_looper.ini file, which when deleted still does not allow me to install a fresh version of the looper.

tofuSCHNITZEL, 2023-12-05:
you cant do any of this from windows. windows can only access the /boot/ part of the sd card and only the ini file is there.
you need to connect to ssh and then just run the https://github.com/adafruit/pi_video_looper#how-to-install commands. (maybe delete the existing pi_video_looper directory first)

Chrisgraue, 2023-12-05:
Yeah, when I try step 2 (git clone), it says that there's already an existing directory and it cannot continue. That's why I figured I'd have to delete stuff before doing that.

tofuSCHNITZEL, 2023-12-06:
yes exactly - you need to delete the existing pi_video_looper directory (but not from windows - directly from your ssh session)
`rm -rf pi_video_looper`

tofuSCHNITZEL, 2023-12-06:
no not at all.. also you are not removing the installation (which you would do with pip) you are just removing the cloned files from the git repo so you can get the fresh ones (and then run install script again which would overwrite the installed module)

if you have done the first install (= the git clone) as the pi user then the folder should be deletable by the pi user - sure you can change to root first and run it then or you use `sudo rm -rf pi_video_looper`

this assumes that you are actually in the folder where the pi_video_looper folder is... usually this is in /home/pi/.

so please check if you are in the correct folder (`ls -lha`) and if you are remove the pi_video_looper folder. run the setup commands from the readme again and then you should be good to go

Chrisgraue, 2023-12-06:
Hmm, `ls -lha` returns a list of a few things:

total 32k
drwxr-xr-x 4 pi pi
drwxr-xr-x 3 root root
-rw------- 1 pi pi ... .bash_history
-rw-r-r--r-- 1 pi pi ... .bash_logout
-rw-r-r--r-- 1 pi pi ... .bashrc
drwx ----- 3 pi pi ... .gnupg
-rw-r-r--r-- 1 pi pi ... .profile
drwxr-xr-x 2 root root ... video

but then when I type `sudo rm -rf pi_video_looper` it just gives me another line. Nothing happens.

I'm trying to get an update working instead of a new image, because I also have a sound card. Every time I've rebuilt the entire image from scratch, it takes me like 6-8 hours of work to get it running again. Although at this point I've been trying to delete files for 2 weeks, so maybe I should just take the L.

Chrisgraue, 2023-12-06:
It may be helpful to note that the way I have found to be able to enter this script was by using windows to delete the video_looper ini file. Otherwise every time I turned it on, it auto-launches the software and I couldn't type in any commands

tofuSCHNITZEL, 2023-12-06:
You should probably do all of this via ssh from the comfort of your own PC over the network, not with a keyboard connected directly to the pi... Also as stated in the readme you can press es on the keyboard to exit the looper, no need to delete the ini file for that. 

according to the ls you sent the folder is already deleted so you should be good to go cloning and reinstalling...

Maybe you should take some time and read up on basic stuff like how to connect to the pi over the network with ssh from your PC and how to list directories delete files etc, basic Linux commands etc... Maybe then also the sound card install will go faster and easier...

tofuSCHNITZEL, 2023-12-06:
Glad it worked out!