adafruit/pi_video_looper #99: No fullscreen background image/color

pi43r, 2020-09-01:
I do not get a fullscreen background, when hitting the  `k` button to restart the video. There is a significant gap of ~100px on every side. My background image, video and monitor have all the same resolution of 1920x1080.

Is there any way to fix this or is it expected behavior?

Running version 1.0.7 on a RPi 3b & Zero W.

tofuSCHNITZEL, 2020-09-04:
have you played around with the overscan settings?
https://raspberrypi.stackexchange.com/questions/23094/calibrate-overscan-settings

pi43r, 2020-09-09:
Well, the video plays in full screen. Wouldn't the overscan setting chop things off at the edges?
I sent out the raspberries as is, so I can not test it anymore. Feel free to close this issue.

closed 2021-12-21 by tofuSCHNITZEL