adafruit/pi_video_looper #100: Using 2 HDMI monitors in Rpi 4

demeteris, 2020-09-03:
Hi,

Thanks for creating a great package. I'm using it with Rpi4 to create a media installation. 
Is there a way to use **pi_video_looper** to show two different videos on two HDMI monitors/TVs? 
For example, there could be two video files (e.g., videopi1 and videopi2) or directories (e.g., video1 and video2)

omxplayer is well capable of doing this. And this thread actually goes into how to do this (playing 2 separate videos on 2 HDMI monitors, on startups).
https://www.raspberrypi.org/forums/viewtopic.php?t=258647

However, I do like certain features of **pi_video_looper*, so I'd rather stick with it for now.
Let me know if this is possible.

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  Raspbian lite

- Python version (run `python -version` or `python3 -version`): Python 3.x (I dont have Rpi on hand. But it's default as I did a fresh installation of Raspbian Lite and pi_video_looper. Nothing else)

- Error message you are receiving, including any Python exception traces:  No error

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): Plug in two HDMI monitors to Rpi4 B. Only one screen shows the video. When "esc" (e.g., in command lines), both screens show the same.

thebackdrifter, 2021-04-21:
Has anyone got this working? I would love to display either a wide video that spans both monitors.

tofuSCHNITZEL, 2021-04-21:
its not that simple unfortunately... the rpi4 rewrite is still in the works and there it is planned to be possible

thebackdrifter, 2021-04-21:
bummer. I have some animated menus I am working on that I would love to have two of the displays connected for a seamless double view. 

Thanks for your response tofu. Appreciate it.

Zindien31, 2022-03-15:
hello, the project would be very useful to me.  anything new on a 2 hdmi 2 different video version?  thank you

tofuSCHNITZEL, 2022-03-15:
Unfortunately not. It's not that easy and I don't have enough time to continue the rework.

gerkylec, 2022-12-13:
У меня работает трансляция на 2 экрана двух разных или одинаковых видео.
Необходимо запустить videolooper вторым потоком со своим конфигом, расположенным по пути /boot/video_looper2.ini
В этом конфиге соответственно прописать параметры запуска чтобы изображение выводилось на второй экран.
Попытаюсь объяснить что нужно сделать
1. Переименовать и заменить файл /usr/local/lib/python3.7/dist-packages/Adafruit_Video_Looper-1.0.7-py3.7.egg
[Adafruit_Video_Looper-1.0.7-py3.7.txt](https://github.com/adafruit/pi_video_looper/files/10216046/Adafruit_Video_Looper-1.0.7-py3.7.txt)
2. Создать файл video_looper2.ini в папке /boot конфиг такой же, только добавить extra_args = --display 7 в файле video_looper.ini - extra_args = --display 2 пути к проигрываемым файлам могут быть разными
3. Добавить в /etc/supervisor/conf.d файл video_looper2.conf с содержимым
 # Supervisord configuration to run video looper at boot and
# ensure it runs continuously.
[program:video_looper2]
command=python3 -u -m Adafruit_Video_Looper2.video_looper
autostart=true
autorestart=unexpected
startsecs=5

По результату должно запускаться примерно так:
htop:
[image]

Проигрываются разные видео на разных hdmi портах

tofuSCHNITZEL, 2023-11-16:
thanks @gerkylec so it IS possible

closed 2023-11-16 by tofuSCHNITZEL