adafruit/pi_video_looper #102: Loop current video forever, press K to go to next video.

JoeReplicant, 2020-09-06:
Hello. Absolutely love this video looper, thank you!

I'm trying to configure the software to seamlessly loop the first video on a flash drive (or playlist) forever. Basically act as if there is only one video on a flash drive. Then if the shortcut K is pressed, it switches to the next video and loops that forever seamlessly. And if pressing K again at the end of a playlist, it just restarts from the top.

Similar to using desktop VLC and selecting "loop one".

This could be useful for many applications, especially signage or info screens that only alternate once a day or once a week to another video. It could be changed by a non-techie person with just the click of a button.

I tried using the _repeat_Nx filename feature with a very high value, but the loop is really not seamless. Having a delay when switching through different videos would be fine for this application, just not when looping the same video. Not sure if I am missing something with the config file. Any help would be highly appreciated.

tofuSCHNITZEL, 2020-09-06:
Hi,
right now "sKip" does not go to the next video but skips the current iteration. so right now there is no way to repeat one file since only if the playlist contains only one file it is looped indefinitely. 
But I understand and like you idea and I see how this would be a good feature to have...
Unfortunately to make this happen quite a bit of reworking is needed to change the play count logic and the skip function.

I will try to look into this in the next couple of days/weeks and I someone is willing to help it would be much appreciated...

JoeReplicant, 2020-09-06:
I wish I could help! Unfortunately I've only just started learning Python.

Could a possible easier workaround in the meantime be to implement some kind of "skip" to the next playlist option? So you can have multiple folders with videos on a flash drive (or in a directory on the SD card) and by pressing a button; it will skip to another folder / playlist. Then it could act just like it does now, where one video in each folder would loop seamlessly. P, for playlist or N for next could be the command. Also seems useful for people with multiple playlists that have multiple videos.

Basically playlists within a playlist.

And maybe an additional option to add something like _hidden to the end of a folder name would not include it when skipping through folders / playlists. The video looper could also constantly monitor a change in the number of folders of a certain directory and reload if detected.

Not sure about the handling of videos on the root of a flash drive (or directory) that aren't in a folder. They could be treated as their own playlist automatically? Then when hitting the command, it could scan through the first folder, and so forth.

Not sure if this is all easier to implement or not, or if a reworking all together would just be better.

tofuSCHNITZEL, 2023-11-16:
maybe this could help in your case:
https://github.com/adafruit/pi_video_looper/issues/195#issuecomment-1669486348
place one video in the folder to loop forever and then switch this video out so the other one gets looped

closed 2023-11-16 by tofuSCHNITZEL