adafruit/pi_video_looper #109: Skewing video on 7" touchscreen

amandansimons, 2020-11-26:
I'm running the video looper on a 7" touchscreen, and the video is squished on the top and bottom only during playback. I am running mp4 files with the omxplayer option. Any tips on how to easily adjust the aspect ratio?

Thanks!

christiansievers, 2020-11-26:
You don't say how your screen is connected but let's assume it's via HDMI. Omxplayer doesn't respect the video output settings you set in config.txt. Instead you’ll have to edit the line in /boot/video_looper.ini that starts with extra_args and add i.e. --win 40, 20, 704, 560 to position the video directly. See the Omxplayer docs for all options: https://github.com/popcornmix/omxplayer

Once you made changes, no need to restart the Pi, just do a sudo supervisorctl restart video_looper to see what’s different.

ltdstudio, 2020-12-25:
Dear Pi_video_looper creater, I hope you can add a command for every video play.
add   
--aspect-mode stretch

especially when I play 480X720 Video on rotated 90° LCD.

tofuSCHNITZEL, 2020-12-26:
you can add this setting to the omx player options in /boot/video_looper.ini

closed 2020-12-26 by tofuSCHNITZEL