adafruit/pi_video_looper #110: Check that adafruit player is running remotely

seanocaster123, 2021-02-06:
I have this app running in a pi remotely so it’s not easy to check if the videos are playing. Currently the only option is phone someone at the location and ask if the video is the tv.

is there any other way of checking myself from the terminal?

tofuSCHNITZEL, 2021-02-06:
I guess you could check htop/top for a omxplayer process... I think they way omxplayer is writing to the videobuffer makes it hard to capture a "screenshot" of the current output.

seanocaster123, 2021-02-06:
> I guess you could check htop/top for a omxplayer process... I think they way omxplayer is writing to the videobuffer makes it hard to capture a "screenshot" of the current output.

Thanks. I tried htop and it shows that 'root' is running Adafruit_Video_Looper with 5-7% CPU and 2% memory in a Raspberry Pi 4b with 2GB memory. Just need to establish if that is how it should be looking.

tofuSCHNITZEL, 2021-02-06:
you should see processes with "omxplayer" (or hello_video) in the name using significant more CPU then the rest of the processes.

seanocaster123, 2021-02-06:
Ok, I did a reboot and now I see the Omxplayer processes and the filename of what is being played. They are at the top of the list and were not there before, I guess that proves that the video was not playing before.

Is there a log that I can follow to try understand why the video player stops playing the playlist?

tofuSCHNITZEL, 2021-02-06:
you can enable console output via the ini - the output will be written to /var/log/supervisor..... (or similar) but I think you should already have output there even without enabling the console output in the ini

seanocaster123, 2021-02-06:
Ok thanks, I've enabled the console and I'm waiting it with tail. I'm also watching htop and for now the 6 videos in the playlist are playing and looping well. Now I just need to keep an eye on it for when the problem occurs and things get stuck.

tofuSCHNITZEL, 2021-02-06:
if you see any crashes or python exceptions please let me know

seanocaster123, 2021-02-06:
Will do. Many thanks for the support.

seanocaster123, 2021-02-07:
Things appear to be running well this morning. Thinking back to yesterday, I had logged into the Raspberry Pi remotely and I had changed the file names of some of the files in the videos folder. I guess that would've confused the playlist and was probably the reason why it stopped playing.

tofuSCHNITZEL, 2021-02-07:
yes this sounds like the reason ;)

seanocaster123, 2021-02-10:
Over the past few days, I've noticed a few things so I'm just trying to troubleshoot them, if I could ask for more of your help please.

1. Is it possible to log the time of each message in the log file? At the moment, the log is working well and I see the file names with 5 second pauses in-between (I've set a .png image to be on screen between videos) but I am noticing that the pi will loop through the 12 videos multiple times without problems, but out of nowhere it will freeze on the .png image for long periods of time before either playing the next video, or not playing it at all.

2. It seems like the log file name changes after every restart - is there a way to prevent this so that it is easier to reopen the terminal windows each time?

Thanks.

tofuSCHNITZEL, 2021-02-10:
Hi,
please try the version from the log_time branch https://github.com/adafruit/pi_video_looper/tree/log_time
it will output a timestamp before each message.

ad 2)
you can add/change "stdout_logfile" in /etc/supervisor/conf.d/video_looper.conf to define a persistant location of the logfile that stores the ouput of the process supervisord is running (see also http://supervisord.org/configuration.html)

seanocaster123, 2021-02-10:
Thanks for the reply. I have implemented the 2nd point successfully but I don't think I'm doing point 1 correctly because I am not seeing the time stamps in the log. I went into the /pi_video_converter/ directory and I ran ` git checkout remotes/origin/log_time` and I got the reply `HEAD is now at 13e60ea add timestamp to output` but I do not see any time stamps when I run `sudo tail -f /var/log/video_looper.log` which is the permanent log file I created

tofuSCHNITZEL, 2021-02-10:
you still have to run `python3 setup.py install --force`

seanocaster123, 2021-02-11:
Thanks, I'm sorry for the ignorance but I'm not very experiences with the git or with the terminal in general. I followed your instructions but it seems the video_looper never runs. From a fresh reboot I ran:

`cd pi_video_looper`
`git checkout remotes/origin/log_time`
`sudo python3 setup.py install --force`

at that point, I didn't see anything happening on htop so I ran`sudo supervisorctl start video_looper` with the result `video_looper: ERROR (spawn error)`

I have now rolled back to `git checkout remotes/origin/HEAD` and things are working again, but without the timestamp still.

Any ideas?

tofuSCHNITZEL, 2021-02-11:
sorry there was an issue in the main file... please check it out again.

seanocaster123, 2021-02-11:
Thanks, I tried the same sequence of steps again I'm still getting this
`video_looper: ERROR (spawn error)`

tofuSCHNITZEL, 2021-02-11:
okay sorry I don't really have time at the moment to test this out properly I just improvised it quickly... what do the logs say?

seanocaster123, 2021-02-11:
Thanks, I appreciate the help and I'm sorry to disturb you.

I'm trying to troubleshoot why the video stops playing. I am not near the pi (I'm 40km away) so I can't physically look at the screen, but from what I notice on htop (cpu usage) and the log file (ie. logging stops), it appears to get stuck on the bg.png image. If I do a 'sudo reboot' then it boots back up and starts playing the playlist and then eventually stops again. This loop happens every time I reboot it and it doesn't seem to stop at the same place every time, so I can't pinpoint the problem. After each reboot I am opening 2 terminal windows, one for htop and one for tailing the log.

Image 1: htop - you can see that even though Omxplayer is running it is not using very much cpu which tells me it is stuck on the image
[https://u.pcloud.link/publink/show?code=XZl4UhXZahlG2bRXQ35CleHKG34lPjSgalNX](url)

Image 2: tail -f /var/log/video_looper.log you can see after 20min it is not adding new entries to the log, which also tells me the video is not playing. (this is the reason I was trying to get the log_time working
[https://u.pcloud.link/publink/show?code=XZE4UhXZjPzpb0vPbsYCP8URk2uKNH6eMMPy](url)

tofuSCHNITZEL, 2021-02-11:
maybe the logfile from running the log_time branch please (the supervisord log file)

seanocaster123, 2021-02-11:
Here is the log taken from /var/log/supervisor/supervisord.log

https://u.pcloud.link/publink/show?code=XZsuUhXZMO9ivrSwlGfEEOBTYwE1IRoN247k

tofuSCHNITZEL, 2021-02-11:
sorry this is not what I meant - but I'm not sure what the correct path is, I dont have a system with me at the moment to check ... but there is a log file somewhere there with "out" or something in the name where everything that is output by the process is written to...

seanocaster123, 2021-02-11:
Thanks, I had a bit of a look around but I'm really out of my league with this stuff. If you manage to remember where I can find such a file, please let me know.

It is evident that something is happening which causes the stream to stop. But it's very weird because after a reboot the videos start and play every time, it's only over time that I start seeing problems.

seanocaster123, 2021-02-11:
Just to point out, from this htop process you can see it has this one video file named '2021-02-09 KBC session 01.mp4' has been running for **0:19.96** which is proof that the player has not moved onto the next file. Have you seen this before?

 1807 root       20   0  187M 27868 21112 S  2.0  1.5  0:19.96 /usr/bin/omxplayer.bin -o both --no-osd --audio_fifo 0.01 --video_fifo 0.01 --align center --font-size 55 /home/pi/videos/2021-02-09 KBC session 01.mp4

tofuSCHNITZEL, 2021-02-11:
if it does not continues it would point to a problem with the video_looper python script because it takes care of starting the next file when the old one is finished

seanocaster123, 2021-02-11:
Thanks. Any suggestions? I'm running Python 2.7.16 and Python 3.7.3

Shall I try remove Adafruit Video player completely and then put it back? Or can I try anything else? Does GPU memory allocation play a part?

tofuSCHNITZEL, 2021-02-11:
most of the time playback issue are because of currupted files with mostly has to do with the SD card getting corrupted so I would suggest to create a complete fresh clean install on a new SD card. if its no potential issue with privacy or copyright - I can offer you that you send me the files and I create a fresh Pi looper here with me and let it run and see what happens...

seanocaster123, 2021-02-11:
Many thanks for that. Would corruptions occur from sudden power offs only, or from sudo reboot commands via ssh as well?

When I originally set it up, it ran for days at my house, looping without any problems. It would be really helpful to understand what went wrong... so please do let me know if you can think of any other way of monitoring the logs in more detail.

I have another Raspberry Pi at home that I'm not using. I will set it up as a new installation and test it using the same videos files.

tofuSCHNITZEL, 2021-02-11:
yes sure sudden poweroffs can cause this... but also just normal "wear" of the sd cards could cause problems when reading the file.. and if omxplayer process hangs the py script will never trigger the playback of the next file - maybe try to get the log_time branch running on your seperate pi (I will try to do it as well)

seanocaster123, 2021-02-11:
Ok thanks. It makes sense but it seems like such bad luck that this would be the reason this time around. I just purchased this pi and 128GB Sandisk microSD card less than a month ago. I'm sure it has had a few sudden power off due to the nature of the location, but I didn't think such things could cause this type of damage. I will try the log_time branch and let you know.

seanocaster123, 2021-02-11:
How should I prevent the pi screen from going to sleep? I remember in the beginning when I had just installed Adafruit player I had a challenge stopping it from blanking. I tried changing setting in autostart etc. - perhaps I did something wrong during that process...

tofuSCHNITZEL, 2021-02-11:
if by pi screen you mean non hdmi - I don't really know. I only used it with screens via HDMI and they never switched off.

seanocaster123, 2021-02-11:
Yes, I meant the HDMI. I was originally having trouble with the screen blanking at random times and I ended up using the Configuration via the GUI to disable blanking. Curious to hear you never needed to disable blanking.

tofuSCHNITZEL, 2021-02-11:
hm weird... with the 10 or so installations I have running with this project (for more than 3 years) never had such issues.

seanocaster123, 2021-02-11:
I am happy to hear that. It hopefully means that I have done something wrong on my end and can resolve it with a fresh install. Will let you know, thanks again for the support.

seanocaster123, 2021-02-12:
Hi, quick update. I have got pi_video_looper working from a new installation on the pi at home and it ran through the night without any problems. I have installed the log_time branch and it is working, see below. I will try a fresh installation on the remote pi and see how it goes.

[2021-02-12 11:41:21.870685] Playing movie: /home/pi/videos/20200713 KBC Promo (1920x1080 Instagram).mp4 (20200713 KBC Promo (1920x1080 Instagram)) 1/1
[2021-02-12 11:44:07.761953] Waiting for: 5 seconds
[2021-02-12 11:44:12.763631] Playing movie: /home/pi/videos/20201223_sean_foil.mp4 (20201223_sean_foil) 1/1
[2021-02-12 11:45:13.891467] Waiting for: 5 seconds
[2021-02-12 11:45:18.897965] Playing movie: /home/pi/videos/2021-02-06 The Diddlebums_smaller.mp4 (2021-02-06 The Diddlebums_smaller) 1/1
[2021-02-12 11:47:02.996104] Waiting for: 5 seconds
[2021-02-12 11:47:08.001481] Playing movie: /home/pi/videos/20210210 Adam surfing at KBC.mp4 (20210210 Adam surfing at KBC) 1/1

seanocaster123, 2021-02-14:
Quick update. The video lopper is running well now after the fresh install. It ran through the night without any hiccups, and the log_time branch is super helpful to make sure things are not stuck! Thanks a lot for the support.

After thinking about it, I think what may have caused the original problem was changes that I made to the display settings in /boot/config.txt - the pi was choosing 4K resolution for my TV so I made changes in that file to force it to use 1920x1080. This is the only thing I did differently the first time around.

tofuSCHNITZEL, 2021-02-14:
great, good to know it works now. please don't forget to disable console output again... otherwise you will run out of storage eventually (since I don't think the logs will be rotated)

closed 2021-02-14 by tofuSCHNITZEL

seanocaster123, 2021-02-28:
Thanks again for the support! Will keep an eye on that logs and the storage. For the moment, things are running 100% reliable and I am very impressed with it. Doing exactly what I need. Amazing piece of software.

tofuSCHNITZEL, 2021-02-28:
great! thanks for the feedback!