adafruit/pi_video_looper #112: Playlist switch

Edwar-Cortes, 2021-03-01:
Hello, I am implementing the video loop, everything works perfectly.

But I want to create a playlist, which I can change at any time and only the videos in the list are played.

Performing a test create the playlist and initialize the video looper, everything works fine. But when I change the playlist it keeps playing the videos from the old playlist.

How can I update the playlist without restarting the video looper service?

tofuSCHNITZEL, 2021-03-01:
Hi,
currently the internal playlist is only rebuilt when the media changes... so e.g. usb drive plugged/unplugged. but also if the filecount in directory mode changes... so a kind of hacky workaround could be that after you change the m3u file you add a new file to the directory (can be empty and I think it does not even need to be a video file)

Edwar-Cortes, 2021-03-01:
I will try it, thank you very much for your prompt reply.

closed 2022-02-14 by tofuSCHNITZEL