adafruit/pi_video_looper #115: How can i save the current state?

Chr-Bur, 2021-05-14:
Dear pi_video_looper team,

at first a big thank you for this script. It works pretty well for me.

i want to ask you for a way to save the current position to start the video which was played last. 
a disabled friend of me couldnt use a dvd player or things like that and it would be a big help for him to watch his favorite soap.

i've tried this in model.py to save the index in a file "wert" and set its value 0 for start:

    def get_next(self, is_random) -> Movie:
        """Get the next movie in the playlist. Will loop to start of playlist
        after reaching end.
        """
        # Check if no movies are in the playlist and return nothing.
        if len(self._movies) == 0:
            return None
        # Start Random movie
        if is_random:
            self._index = random.randrange(0, self.length())
        else:
            # Start at the first movie and increment through them in order.
            if self._index is None:
                with open('Wert.txt','r')as f:
                    c = int(f.read())
                self._index = c
            else:
                self._index += 1
                with open('Wert.txt','w')as f:
                    f.write(str(self._index))
            # Wrap around to the start after finishing.
            if self._index >= self.length():
                self._index = 0
                with open('Wert.txt','w')as f:
                    f.write(str(self._index))

        return self._movies[self._index]

now the looper crashes after countdown, any ideas how to fix this?

sorry for my bad english

tofuSCHNITZEL, 2021-05-14:
I guess because on the first run this file does not exist so it throws a FileNotFoundException.
But anyway, I created a new branch with this feature: https://github.com/adafruit/pi_video_looper/tree/playlist_resume
there is also a new setting in the ini. you need to set resume_playlist = true.

disclaimer:
unfortunately I had no time to test it on real hardware - so please pull the new branch and try it out for yourself ;)
if should be "stable" in terms of index file not existing and also if the number of files changes (to less) than the index that was previously saved - in this case it would start at the beginning. but in case of more files it would still resume from the last index.

Chr-Bur, 2021-05-14:
Thank you for your fast support. Thats awesome!

i've tried the new branch and it doesn't work either. 
It starts playing with the first movie all the time but at least it start again👍 
 
i've set resume_playlist=true after installation and than sudo ./reload . thats right isnt it? 

i just unplug the pi maybe thats part of the problem? he isnt able to do more so this is really imortant thing.

tofuSCHNITZEL, 2021-05-14:
okay maybe the problem is that the file cant get written to...
the reload command is correct.
the index state is written when the next movie plays. so you have to wait until file 2 plays then if you unplug it and plug it back in (or reload the player) it should start with file 2.
I will try to install it on hardware later if I have time.

tofuSCHNITZEL, 2021-05-14:
quick update: just tried it on my rpi and it works like intended... you can check the file /playlist_index.txt via ssh to see that the counter is increased after a video is played.

Chr-Bur, 2021-05-14:
ok thats strange. 
i downloaded the new branch and installed everything new.
 new raspian lite and then winscp to send the right branch to the pi(then chmod a+x install.sh and sudo ./install.sh) , 
 i installed it an it startet without save.
 after that i edited the ini and set resume=true. 
then chmod a+x reload.sh and sudo ./reload. 

the usb stick has 6 examplevideos idk like 10 secs long all mp4 

the txt isnt there when i check over ssh, 
i make one by my own and write 2 in (just for test) 
it wont work and i cant even read the txt after that (tried chmod 777 but nothing)

Chr-Bur, 2021-05-28:
I have tryed it the "usual" way and pulled the playlist_resume-branch, but it doesnt work either.
Any ideas how i could fix this? 
there is no playlist_index.txt, i create one by myself but it does not help.

tofuSCHNITZEL, 2023-11-16:
Stale - please comment if this is still a problem

closed 2023-11-16 by tofuSCHNITZEL