adafruit/pi_video_looper #116: Audio files

simonemarin, 2021-06-03:
Hallo aren't audio files supported? Since it is often used in media installation that would be an easy requirement. Cheers.

tofuSCHNITZEL, 2021-06-03:
You can just put your audio in a blank video file. But the audio output of the rpi is quite unusable, so you would need an extra audio interface or hdmi audio extractor anyway.
you can also try to enable mp3 files via the ini and test if omxplayer plays them even if they don't have video.

closed 2021-06-03 by tofuSCHNITZEL

simonemarin, 2021-06-03:
Hey, thanks for the fast reply, yes indeed both MP3 and WAV files do work, with MP3 you even get good sound quality (form the audio jack), WAV is ok, better than video.

simonemarin, 2021-06-03:
Actually WAV files are also ok, it was my sound file which had some artifacts.