adafruit/pi_video_looper #126: Playback Stops and Screen Goes Black After Some Time

jhnewton84, 2021-10-05:
I've currently got a RPI3 setup to loop about 300 video files, around 30GB of SD content. I'm running only Pi Video Looper on the latest' version of Raspberry Pi OS Lite. OS is on a SanDisk Ultra 32GB MicroSD and the videos are playing on a 128GB Verbatim flash drive. I'm using the CanaKit official power supply and no other peripherals are connected. I'm outputting to an old CRT TV in a spare bedroom. All videos are in M4V, encoded using Handbrake in a batch session.

All that being said, everything seems to work fine, the videos begin to loop without issue, but when I come back to check on it after several hours the loop has stopped and the screen is black. If I remove and replace the flash drive, the loop starts over. I've tried with two different flash drives, same result.

Any idea what could be going on here? Is there something causing it to timeout? A log file that could help me out?

I'm pretty green when it comes to Linux, Python, etc., so I'm a little annoyed that all conditions seem to be correct but something is failing here. Hopefully someone here can help me out. Much appreciated!

tofuSCHNITZEL, 2021-10-05:
Hi,
please set "console_output = true" in /boot/video_looper.ini and then let it run and if it happens look in the log files located in /var/log/supervisord/* (and post them here)

jhnewton84, 2021-10-05:
Okay, cool. I changed the .ini file and now we wait. I'll check back in when I get the results. Thank you.

jhnewton84, 2021-10-06:
@tofuSCHNITZEL Uploaded the log file [here](https://www.dropbox.com/s/03lg7enargijb5q/supervisord.log?dl=1). Hope this helps. Thanks again!

tofuSCHNITZEL, 2021-10-06:
great! glad to hear!